ccMSC/ckb #103: IT layout

Vortigern, 2015-08-12:
Dear MSC,
I've just bought a K70 RGB: everything is working fine with your daemon.
The only thing I miss is the Italian layout. Are you going to introduce it at some point?
Is there something I could do to the help?

Vortigern

ccMSC, 2015-08-13:
Added in 8a8ea170613b8d2e6b2c753abefed9ccfc2daced. Let me know if everything looks OK.

Vortigern, 2015-08-13:
Hi,
thank you very much for adding the IT layout!
The only thing that's wrong is in the  src/ckb/keymap.cpp  row 51 {0, "¡", "equal"}
Actually it's not "i" but "ì" with the  accent. So it should looks like:
```
{0, "\\", "grave"}, {0, "'", "minus"}, {0, "ì", "equal"},
```
Thanks again for your work!
Vortigern

ccMSC, 2015-08-14:
Ahh my bad, I read it wrong. Fixed now.

Vortigern, 2015-08-14:
Perfect! Now it's ok!
Thanks again!

Vortigern

closed 2015-08-14 by Vortigern