ccMSC/ckb #105: M65 RGB. Scroll wheel

sandypickle, 2015-08-13:
I downloaded the program. It all worked pretty well until I came a cross a problem. My scroll wheel. No matter what I do, scrolling up and down doesn't work. It doesn't do anything. For me, scrolling is a big part of my computer usage. I am constantly scrolling. I have a Mac computer. I uninstalled ckb and restarted my computer. All of a sudden it's working again. Then, I reinstall the program and it stops working. Is there ay solution to this? Thank you.

ccMSC, 2015-08-14:
Which OSX version? Have you tried changing the scroll speed in System Preferences?

sandypickle, 2015-08-14:
I have 10.9.5. And when I have CKB installed, the mouse isn't detected.

ccMSC, 2015-08-15:
I may have found the issue. It looks like the OS would think there was no mouse connected sometimes. See if the latest testing version works for you.

If you need a binary package: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2%2Bt04.zip

sandypickle, 2015-08-15:
THANK YOU THANK YOU THANK YOU! After getting the RGB mouse, the thought of not being able to change colors made me so mad. You sir are a life saver. I quickly installed it and this new testing version didn't interfere with ANYTHING. Also the interface of this version isn't that much different, but there a couple things that have ben changed that makes it so much easier. The scrolling is great and the computer detects the mouse. I wish I could hug you in person. Thank you so much

ccMSC, 2015-08-18:
No problem :)

I'll leave this ticket open until the changes are put back in the main version, probably by the end of the week.

closed 2015-08-26 by ccMSC