ccMSC/ckb #106: OSX upgrade to 10.10.5 breaks CKB

linjer, 2015-08-14:
Today's release of the 10.10.5 OSX update seems to break CKB. I upgraded via app store, rebooted, and now CKB cannot detect my K70 RGB. I've tried moving the polling rate/bios switch on the keyboard, unplugging/replugging, killing CKB, and various combinations. Still shows up as "no devices detected", although I can type just fine.

linjer, 2015-08-14:
Just as an update, I got things back to normal but it took me noticing my computer fan was running on full speed to see the ckb-daemon was constantly eating up 52.8% of my CPU, even when I unplugged the keyboard and plugged it back in.

Things returned to normal when quit out of the CKB front end, I did `sudo kill -9 96` which was the pid for the ckb-daemon, then plugged the keyboard back in. Where can I find logs for what was going on?

ccMSC, 2015-08-15:
Strange, still works normally on my computer.

The easiest way to get the daemon output is to stop the system service and run it manually. In Terminal:
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```
Recommend you type these with a separate keyboard if possible. If not, you may need to slide the poll rate switch back and forth to stop the terminal from getting spammed with enter keys.

If the issue doesn't occur when running manually, you can edit the file `/Library/LaunchDaemons/com.ckb.daemon.plist`:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Label</key>
    <string>com.ckb.daemon</string>
    <key>ProgramArguments</key>
    <array>
        <string>/Applications/ckb.app/Contents/Resources/ckb-daemon</string>
    </array>
    <key>KeepAlive</key>
    <true/>
    <key>ProcessType</key>
    <string>Interactive</string>
    <key>LegacyTimers</key>
    <true/>
    <key>StandardOutPath</key>
    <string>/var/log/ckb.log</string>
    <key>StandardErrorPath</key>
    <string>/var/log/ckb.err.log</string>
</dict>
</plist>
```
Then reboot. This should produce two log files, `/var/log/ckb.log` and `/var/log/ckb.err.log` which contain the driver output. IIRC you have to stop the daemon (the `sudo launchctl unload` line) in order to get the full output.

linjer, 2015-08-17:
I've added the logging entries to the plist. I haven't encountered this issue since filing, but I use the K70 in the office and the laptop as-is at home, so perhaps the constant back and forth may induce the situation again.

For now, it seems it's not the update itself that is culprit, so maybe it's best you close this issue. I'll raise a new one once I have the logs and the scenario reproduced.

closed 2015-08-18 by ccMSC