ccMSC/ckb #107: Polish layout

sniewczas, 2015-08-17:
Hi. Could you please add the [Polish programmer's](https://en.wikipedia.org/wiki/QWERTY#Polish) layout? That would be awesome.

ccMSC, 2015-08-19:
Added in f1b98c80b8c659c5479fae3dbb2cb1a32c725eb2.

sniewczas, 2015-08-19:
Now I'm not exactly sure if creating this issue was neccesary, because I just logged in to try the f1b98c8 and noticed that polish characters work just fine on v0.2 release. I couldn't get them to work previously, the layout was set to United Kingdom in cqb and there was no Polish layout available on the list, so I figured that cqb doesn't support it and decided to open this issue...

I uninstalled the 0.2 and installed 0.2+t06 and it didn't work again. I removed ckb completely and it didn't work either, so I went to check my OS keyboard settings. The only layout in Cinnamon keyboard settings was Polish, so I added another one, random, just so I can switch between them, and it works now. So it looks like there is something wrong with my OS (Mint 17.1, Cinnamon) which makes it forget the set layout sometimes. I'm on 0.2+t06 now and everything's just fine. Anyway, thanks for your time!

closed 2015-08-19 by sniewczas