ccMSC/ckb #109: CKB daemon using high CPU and cannot detect keyboard

linjer, 2015-08-17:
In reference to my previous issue filed, here is a screenshot of what I was experiencing. CKB at ~52.8% load, and it cannot detect the keyboard. After killing the daemon process, it comes back up and operates as expected.

Here is the dumb from the logs:
ckb.log
```
    ckb: Corsair RGB driver beta-v0.2
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: 0902001CAE3D8C8153A90E67F5001947)
[I] Starting input thread for /var/run/ckb1
[I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1

[I] Caught signal 15
[I] Disconnecting Corsair K70 RGB Gaming Keyboard (S/N: 0902001CAE3D8C8153A90E67F5001947)
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb0
```

ckb.err.log
```
[E] iterate_devices (usb_mac.c:322): Failed to create device plugin: e00002c7
[E] iterate_devices (usb_mac.c:322): Failed to create device plugin: e00002c7
```

http://i.imgur.com/TiR9goz.png

ccMSC, 2015-08-18:
So the error it's throwing up is `kIOReturnUnsupported` but I can't seem to find any explanation as to why it might do that or why it works sometimes but not others. I noticed a handful of small issues in the code around that area - none of which should cause problems according to Apple's documentation, but it's pretty incomplete, so it's possible that they're related anyway.

You can find the latest changes in the testing repository, or a binary package here: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2%2Bt05.zip
Keep in mind you'll have to re-edit `com.ckb.daemon.plist` after installing. Let me know if you still get the errors or not.

linjer, 2015-08-18:
Perhaps we can increase details in the logging, along with timestamps? If you don't think that'll be helpful, I'll install the binary linked above.

linjer, 2015-08-18:
And thanks for looking into this.

linjer, 2015-08-19:
I haven't installed your new binary yet, but I have a different, but possibly related issue. The CKB frontend app is now at 100% cpu load (`ckb-daemon` appears to be normal).

http://imgur.com/a/RLEhR

Here is the log contents. It contains entries which were previously reported above as well.

```
[jerrylin@descartes ~ ]$ cat /var/log/ckb.log
    ckb: Corsair RGB driver beta-v0.2
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: 0902001CAE3D8C8153A90E67F5001947)
[I] Starting input thread for /var/run/ckb1
[I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1

[I] Caught signal 15
[I] Disconnecting Corsair K70 RGB Gaming Keyboard (S/N: 0902001CAE3D8C8153A90E67F5001947)
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb0
```

```
[jerrylin@descartes ~ ]$ cat /var/log/ckb.err.log
[E] iterate_devices (usb_mac.c:322): Failed to create device plugin: e00002c7
[E] iterate_devices (usb_mac.c:322): Failed to create device plugin: e00002c7
[E] os_usbsend (via led_keyboard.c:118): Got return value 0xe00002c2
[E] usb_tryreset (usb.c:175): Reset failed. Disconnecting.
```

ccMSC, 2015-08-19:
There is an OSX-specific issue where the GUI can get locked up if the driver spends a long time getting the device connected. It's probably that. I don't think more detailed logging would be possible in this instance since `IOCreatePlugInInterfaceForService` is actually the first thing that it does.

linjer, 2015-08-19:
Just installed the binary you linked above. Will try to remember to report back within a week. If no issues by then I guess we'll close this out until I experience something again and/or I gain new information.

RobErskine, 2015-08-25:
I'm experiencing this same issue as well. I installed ckb initially and it worked great. I restarted my machine and noticed that ckb could no longer find my keyboard, and from there the cpu would rise. I will also try and install the mentioned release above and let you know how I make out.

linjer, 2015-08-25:
@RobErskine , reporting in after 6 days of use with the posted binary, and i haven't had a single issue. Before, it seemed to happen maybe 2 times a week, but complete guesstimate.

ccMSC, 2015-08-26:
Fix merged in v0.2.1. @RobErskine if you still experience the issue, can you post a log like @linjer 's?

linjer, 2015-08-31:
I've updated to v0.2.1 and haven't seen issues yet. Closing this out and will open a new ticket if something arises.

closed 2015-08-31 by linjer