ccMSC/ckb #112: ckb icon has a white background in notification area

thomassross, 2015-08-23:
For some reason the ckb icon has a white background in my notification area:
[image]

ccMSC, 2015-08-24:
Which OS/desktop environment? If you're using Unity, did you install libappindicator before building?

thomassross, 2015-08-24:
Debian Testing, with XFCE 4.12

ccMSC, 2015-08-26:
Tried with XFCE on Arch, not getting this issue. It's entirely possible that this is another bug in QTrayIcon (I already know of several), which unfortunately means there's nothing I can do to fix it. I may try to get rid of QTrayIcon entirely at some point and replace it with something more functional, but I can't promise anything yet.

thomassross, 2015-08-26:
I found a fix, enabling window manager compositing and restarting ckb fixed it as described [here](https://github.com/telegramdesktop/tdesktop/issues/874)

closed 2015-08-26 by thomassross