ccMSC/ckb #114: Extreme CPU usage when hotplugging the keyboard with a KVM

Thalagyrt, 2015-08-25:
Hiya,

On two Macs running OSX 10.10.5, I'm seeing extreme CPU usage after a day's work. I'm running a KVM that essentially unplugs the keyboard from one computer and plugs it into the other when I switch machines, so ckb is seeing the keyboard get detached and reattached quite a bit. At the end of the day after a lot of switches, the fans are noticeably loud and ckb's spinning 2-3 full CPU cores on both machines. It seems that perhaps there might be a loop somewhere that doesn't get cleaned up when the keyboard is unplugged?

I haven't been able to 100% pinpoint the scenario that causes this behavior. I've swapped back and forth and I've noticed the thread count in ckb is gradually rising as I swap. Not necessarily 1 thread per swap, but 1 thread every few swaps. In the most recent swap, the thread count incremented by one and now there's a thread eating an entire CPU core. I've attached a sample I yanked from activity monitor while this was happening - not sure if it'll be helpful or not, but can't hurt to have the info. If there's any way for me to get you better info, please let me know! :)

Thanks!
-James

```
Sampling process 60959 for 3 seconds with 1 millisecond of run time between samples
Sampling completed, processing symbols...
Analysis of sampling ckb (pid 60959) every 1 millisecond
Process:         ckb [60959]
Path:            /Applications/ckb.app/Contents/MacOS/ckb
Load Address:    0x1053bf000
Identifier:      ckb
Version:         0
Code Type:       X86-64
Parent Process:  ??? [1]

Date/Time:       2015-08-24 23:20:32.371 -0400
OS Version:      Mac OS X 10.10.4 (14E46)
Report Version:  7
Analysis Tool:   /usr/bin/sample
----

Call graph:
    2723 Thread_4522243   DispatchQueue_1: com.apple.main-thread  (serial)
    + 2723 start  (in libdyld.dylib) + 1  [0x7fff8c47b5c9]
    +   2723 main  (in ckb) + 1047  [0x1053c3b07]
    +     2723 QCoreApplication::exec()  (in QtCore) + 346  [0x1062c178a]
    +       2723 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)  (in QtCore) + 381  [0x1062be8cd]
    +         2723 ???  (in libqcocoa.dylib)  load address 0x109212000 + 0x217fd  [0x1092337fd]
    +           2723 -[NSApplication run]  (in AppKit) + 594  [0x7fff84bb1af3]
    +             2723 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:]  (in AppKit) + 346  [0x7fff84bbbe58]
    +               2723 _DPSNextEvent  (in AppKit) + 978  [0x7fff84bbc8ab]
    +                 2723 _BlockUntilNextEventMatchingListInModeWithFilter  (in HIToolbox) + 71  [0x7fff8de0812b]
    +                   2722 ReceiveNextEventCommon  (in HIToolbox) + 431  [0x7fff8de082ea]
    +                   ! 2719 RunCurrentEventLoopInMode  (in HIToolbox) + 235  [0x7fff8de0856f]
    +                   ! : 2717 CFRunLoopRunSpecific  (in CoreFoundation) + 296  [0x7fff8e9bcbd8]
    +                   ! : | 2491 __CFRunLoopRun  (in CoreFoundation) + 1371  [0x7fff8e9bd37b]
    +                   ! : | + 2491 __CFRunLoopServiceMachPort  (in CoreFoundation) + 212  [0x7fff8e9bdeb4]
    +                   ! : | +   2491 mach_msg  (in libsystem_kernel.dylib) + 55  [0x7fff9032964f]
    +                   ! : | +     2491 mach_msg_trap  (in libsystem_kernel.dylib) + 10  [0x7fff9032a4de]
    +                   ! : | 207 __CFRunLoopRun  (in CoreFoundation) + 927  [0x7fff8e9bd1bf]
    +                   ! : | + 180 __CFRunLoopDoSources0  (in CoreFoundation) + 269  [0x7fff8e9bdb8d]
    +                   ! : | + ! 180 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__  (in CoreFoundation) + 17  [0x7fff8e9cba01]
    +                   ! : | + !   131 ???  (in libqcocoa.dylib)  load address 0x109212000 + 0x20be2  [0x109232be2]
    +                   ! : | + !   : 131 QTimerInfoList::activateTimers()  (in QtCore) + 1270  [0x1063130c6]
    +                   ! : | + !   :   131 QCoreApplication::notifyInternal(QObject*, QEvent*)  (in QtCore) + 115  [0x1062c11b3]
    +                   ! : | + !   :     131 QApplication::notify(QObject*, QEvent*)  (in QtWidgets) + 8318  [0x105553ade]
    +                   ! : | + !   :       131 QApplicationPrivate::notify_helper(QObject*, QEvent*)  (in QtWidgets) + 251  [0x1055506fb]
    +                   ! : | + !   :         131 QObject::event(QEvent*)  (in QtCore) + 48  [0x1062ea490]
    +                   ! : | + !   :           130 QMetaObject::activate(QObject*, int, int, void**)  (in QtCore) + 2994  [0x1062f1b82]
    +                   ! : | + !   :           | 78 Kb::frameUpdate()  (in ckb) + 28  [0x1054241ac]
    +                   ! : | + !   :           | + 46 getMuteState  (in ckb) + 149  [0x1053c3695]
    +                   ! : | + !   :           | + ! 42 AudioHardwareServiceGetPropertyData  (in AudioToolbox) + 1164  [0x7fff8f186cb1]
    +                   ! : | + !   :           | + ! : 42 Client_AHS_GetPropertyData  (in AudioToolbox) + 156  [0x7fff8f185c89]
    +                   ! : | + !   :           | + ! :   42 mach_msg  (in libsystem_kernel.dylib) + 55  [0x7fff9032964f]
    +                   ! : | + !   :           | + ! :     42 mach_msg_trap  (in libsystem_kernel.dylib) + 10  [0x7fff9032a4de]
    +                   ! : | + !   :           | + ! 2 AudioHardwareServiceGetPropertyData  (in AudioToolbox) + 84  [0x7fff8f186879]
    +                   ! : | + !   :           | + ! : 2 AudioHardwareServiceHasProperty  (in AudioToolbox) + 61  [0x7fff8f186403]
    +                   ! : | + !   :           | + ! :   2 Client_AHS_HasProperty  (in AudioToolbox) + 129  [0x7fff8f185a60]
    +                   ! : | + !   :           | + ! :     2 mach_msg  (in libsystem_kernel.dylib) + 55  [0x7fff9032964f]
    +                   ! : | + !   :           | + ! :       2 mach_msg_trap  (in libsystem_kernel.dylib) + 10  [0x7fff9032a4de]
    +                   ! : | + !   :           | + ! 2 AudioHardwareServiceGetPropertyData  (in AudioToolbox) + 1289  [0x7fff8f186d2e]
    +                   ! : | + !   :           | + !   2 mach_vm_deallocate  (in libsystem_kernel.dylib) + 25  [0x7fff9032e153]
    +                   ! : | + !   :           | + !     2 _kernelrpc_mach_vm_deallocate_trap  (in libsystem_kernel.dylib) + 10  [0x7fff9032a412]
    +                   ! : | + !   :           | + 25 getMuteState  (in ckb) + 69  [0x1053c3645]
[... 1008 lines cut ...]
    0x7fff8f89b000 -     0x7fff8f9cbfff  com.apple.UIFoundation (1.0 - 1) <466BDFA8-0B9F-3AB0-989D-F9779422926A> /System/Library/PrivateFrameworks/UIFoundation.framework/Versions/A/UIFoundation
    0x7fff8f9cc000 -     0x7fff8f9fcfff  com.apple.GSS (4.0 - 2.0) <7DE487D3-50C8-3A45-8C17-B8806B5E6AA5> /System/Library/Frameworks/GSS.framework/Versions/A/GSS
    0x7fff8f9fd000 -     0x7fff8fa00fff  com.apple.help (1.3.3 - 46) <CA4541F4-CEF5-355C-8F1F-EA65DC1B400F> /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/Help.framework/Versions/A/Help
    0x7fff8fa30000 -     0x7fff8fa30fff  com.apple.Accelerate (1.10 - Accelerate 1.10) <2C8AF258-4F11-3BEC-A826-22D7199B3975> /System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate
    0x7fff8fa31000 -     0x7fff8fa31ff7  liblaunch.dylib (559.30.1) <B1301610-D60C-3301-B254-11F066BD48A7> /usr/lib/system/liblaunch.dylib
    0x7fff90035000 -     0x7fff90037fff  libCVMSPluginSupport.dylib (11.1.2) <1C5C1757-67F1-3C23-90EF-643619A0E7DC> /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libCVMSPluginSupport.dylib
    0x7fff90098000 -     0x7fff9029246f  libobjc.A.dylib (647) <759E155D-BC42-3D4E-869B-6F57D477177C> /usr/lib/libobjc.A.dylib
    0x7fff90319000 -     0x7fff90336fff  libsystem_kernel.dylib (2782.30.5) <101D28C0-AF07-3B81-87BE-CA27ADED33AB> /usr/lib/system/libsystem_kernel.dylib
    0x7fff90367000 -     0x7fff9036ffe7  libcldcpuengine.dylib (2.4.5) <6CC680F9-0D13-333B-B151-F0D9C631A1E5> /System/Library/Frameworks/OpenCL.framework/Versions/A/Libraries/libcldcpuengine.dylib
    0x7fff909da000 -     0x7fff90a39fff  com.apple.AE (681.2 - 681.2) <450F45DB-0F60-383D-BD22-03E296C82675> /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/AE.framework/Versions/A/AE
    0x7fff90cdb000 -     0x7fff90ce4fff  libsystem_pthread.dylib (105.10.1) <3103AA7F-3BAE-3673-9649-47FFD7E15C97> /usr/lib/system/libsystem_pthread.dylib
    0x7fff90d2f000 -     0x7fff90d2ffff  libOpenScriptingUtil.dylib (162.2) <D6A2216D-ADB2-3F24-AD30-F6D00829F545> /usr/lib/libOpenScriptingUtil.dylib
[... cut ...]
    0x7fff91160000 -     0x7fff911a6ff7  libFontRegistry.dylib (134.1) <CE41D8C2-BEED-345C-BC4F-3775CC06C672> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/ATS.framework/Versions/A/Resources/libFontRegistry.dylib
    0x7fff911e5000 -     0x7fff911ebff7  libsystem_networkextension.dylib (167.30.1) <3E99FF35-DCBB-3A4C-8853-F1F39A792D29> /usr/lib/system/libsystem_networkextension.dylib
    0x7fff911ec000 -     0x7fff91213fff  com.apple.shortcut (2.14 - 2.14) <0E9228EC-E688-3E83-9516-5211FFEA923E> /System/Library/PrivateFrameworks/Shortcut.framework/Versions/A/Shortcut
Sample analysis of process 60959 written to file /dev/stdout
```

ccMSC, 2015-08-26:
Possibly related to #109? I tried replicating the problem by unplugging and replugging the keyboard a few times but so far nothing unusual has happened. Can you narrow down whether it's possible to recreate that way, or does it only happen with the KVM?

Thalagyrt, 2015-08-26:
Heh, it hasn't happened once today, despite easily a hundred switches back and forth. Seems to be really temperamental - I'll update you if I figure out a way to repro it!

Thalagyrt, 2015-08-31:
I haven't been able to replicate the issue when plugging/unplugging the keyboard, however it still happens a several times a day when switching back and forth with the KVM. Is there anything you'd like me to try to do to get you more information on the process state when it happens?

I think it's related to the keyboard being initialized and running when CKB gets it as a new device, though I might be wrong there. The keyboard doesn't have any time to power down, and it still retains the old profile from the old computer for about 10 seconds before CKB on the new computer takes over.

ccMSC, 2015-09-01:
To clarify, this is `ckb` that's using a lot of CPU and not `ckb-daemon`?

I'm wondering if the driver/OS is seeing multiple keyboards connected. Try doing an `ls /var/run | grep ckb` and see if there's more than one device there. If so, can you give me the output of `cat /var/run/ckbX/features` for each of them, where `X` is the device number?

Thalagyrt, 2015-09-01:
Yeah, it's the `ckb` process. As soon as it does it again I'll get you that info! Thanks! :+1:

Thalagyrt, 2015-09-01:
Also worth mentioning, this is when it's not doing it. I do have another "keyboard" attached since the KVM does keyboard emulation:

```
eldenser:~ jriley$ ls /var/run | grep ckb
ckb0
ckb1
eldenser:~ jriley$ cat /var/run/ckb0/features
cat: /var/run/ckb0/features: No such file or directory
eldenser:~ jriley$ cat /var/run/ckb1/features
corsair k70 rgb pollrate bind notify fwversion fwupdate
```

Thalagyrt, 2015-09-01:
Flipped back and forth until it happened on the other computer.

```
olothontor:~ jriley$ ls /var/run | grep ckb
ckb0
ckb1
olothontor:~ jriley$ cat /var/run/ckb0/features
cat: /var/run/ckb0/features: No such file or directory
olothontor:~ jriley$ cat /var/run/ckb1/features
corsair k70 rgb pollrate bind notify fwversion fwupdate
```

Still just two devices, one Corsair and one emulated keyboard.

Thalagyrt, 2015-09-01:
Open files when it's misbehaving:

```
/Applications
/Applications/ckb.app/Contents/MacOS/ckb
/Applications/ckb.app/Contents/Frameworks/QtWidgets.framework/Versions/5/QtWidgets
/Applications/ckb.app/Contents/Frameworks/QtGui.framework/Versions/5/QtGui
/Applications/ckb.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore
/Applications/ckb.app/Contents/Frameworks/QtNetwork.framework/Versions/5/QtNetwork
/System/Library/Frameworks/AGL.framework/Versions/A/AGL
/System/Library/ColorSync/Profiles/sRGB Profile.icc
/usr/share/icu/icudt53l.dat
/Applications/ckb.app/Contents/PlugIns/platforms/libqcocoa.dylib
/Applications/ckb.app/Contents/Frameworks/QtDBus.framework/Versions/5/QtDBus
/Applications/ckb.app/Contents/Frameworks/QtPrintSupport.framework/Versions/5/QtPrintSupport
/System/Library/ColorSync/Profiles/Generic RGB Profile.icc
/System/Library/ColorSync/Profiles/Generic Gray Gamma 2.2 Profile.icc
/System/Library/ColorSync/Profiles/Generic Gray Profile.icc
/System/Library/Fonts/Helvetica.dfont
/private/var/folders/zg/ndfy6_dj78d371qzv674cb9c0000gp/0/com.apple.LaunchServices-107502.csstore
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/SystemAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/GraphiteAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityGraphiteAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/DarkAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/GraphiteDarkAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityDarkAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityDarkGraphiteAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/VibrantLightAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityVibrantLightAppearance.car
/System/Library/PrivateFrameworks/CoreUI.framework/Versions/A/Resources/SArtFile.bin
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/Resources/Extras2.rsrc
/System/Library/Fonts/HelveticaNeueDeskInterface.ttc
/System/Library/Fonts/Monaco.dfont
/Applications/ckb.app/Contents/PlugIns/imageformats/libqdds.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqgif.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqicns.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqico.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqjp2.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqjpeg.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqmng.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqtga.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqtiff.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqwbmp.dylib
/Applications/ckb.app/Contents/PlugIns/imageformats/libqwebp.dylib
/System/Library/Fonts/LucidaGrande.ttc
/System/Library/Fonts/GeezaPro.ttc
/System/Library/Fonts/Thonburi.ttc
/Library/Fonts/Kailasa.ttf
/System/Library/Fonts/STHeiti Light.ttc
/System/Library/Fonts/STHeiti Medium.ttc
/System/Library/Keyboard Layouts/AppleKeyboardLayouts.bundle/Contents/Resources/AppleKeyboardLayouts-L.dat
/System/Library/Caches/com.apple.IntlDataCache.le.kbdx
/usr/lib/dyld
/private/var/db/dyld/dyld_shared_cache_x86_64h
/dev/null
/dev/null
/dev/null
count=2, state=0x2
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/SystemAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/GraphiteAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityGraphiteAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/DarkAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/GraphiteDarkAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityDarkAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityDarkGraphiteAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/VibrantLightAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityVibrantLightAppearance.car
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/Resources/Extras2.rsrc
[... 117 lines cut ...]
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityDarkGraphiteAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/VibrantLightAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/AccessibilityVibrantLightAppearance.car
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/Resources/Extras2.rsrc
->0x70a0221b461f9195
->0x70a0221b58b79765
/private/var/run/ckb1/notify1
/private/var/run/ckb1/cmd
->0x70a0221b3993d8e5
->0x70a0221b460d6115
->0x70a0221b461f8065
->0x70a0221b3476ea85
->0x70a0221b58b79b85
->0x70a0221b461f8e25
->0x70a0221b461f9df5
->0x70a0221b58b78dc5
->0x70a0221b58b794a5
->0x70a0221b461faa55
->0x70a0221b58b78635
->0x70a0221b461f9455
->0x70a0221b3476cfb5
->0x70a0221b461fb765
```

Looks like misbehaving it has a bunch of handles to notify1, and when it's behaving properly it doesn't. Does that help? :)

ccMSC, 2015-09-01:
Ok, looks like the problem is that the notify-reading thread isn't being closed for some reason. I'll look into it and let you know when I find something.

Thalagyrt, 2015-09-01:
Alright. If there's anything else you'd like me to do to get more info/test for you let me know - I'm pretty much on GitHub all day anyway.

ccMSC, 2015-09-08:
Ok, my working theory is that the KVM is causing the device to disconnect and then reconnect rapidly at some point. The thread stays open because the GUI asks the daemon to re-create the notification node so quickly that it doesn't have a chance to shut down the old node first. Now, ckb actually does try to terminate the thread after closing a device, but it looks like that's not happening.

Anyway, if this is indeed the case then it should be fixed by 98366fd247bb3bb7a0a363b0e0dea688069a1680. If that doesn't do the trick then I'll look into other causes.

Thalagyrt, 2015-09-08:
I'm compiling this now - I'll let you know if I see an improvement.

Thalagyrt, 2015-09-08:
So far so good! I'll post another update in a few days.

Thalagyrt, 2015-09-09:
Yep, that definitely fixed the issue. I'll close this out so it's out of your hair. Thanks so much!

closed 2015-09-09 by Thalagyrt