ccMSC/ckb #117: Mac OS X - certain keys type odd characters

conrad-vanl, 2015-08-31:
This has been going on for some time now, but just upgraded to latest beta and confirm it's still an issue for me. 

Using a K95 RGB fyi.

When CKB is opened, several keys will start to insert odd characters instead of what they should. For example, the 'tilde' button inserts § and ± (with shift key held). 

If I cycle the bios switch, the keys return to their normal operation momentarily, but then start inserting odd characters again. If I exit CKB, then cycle the bios switch, the keys return to their normal operation and stay. But as soon as I open CKB again, the issue creeps back up.

DonutEspresso, 2015-08-31:
Check your keyboard layout settings in CKB - it's likely you might have UK selected instead of US layout.

conrad-vanl, 2015-09-01:
Yup, wow can't believe I didn't notice that. Thanks!

closed 2015-09-01 by conrad-vanl