ccMSC/ckb #118: CKB high CPU use

mxanth, 2015-09-05:
Was seeing excess CPU usage with update. Unchecking the "show animated" box on keyboard layout process CPU usage dropped dramatically. Not sure if this is unique to me but looks like a temporary fix.

mxanth, 2015-09-05:
Same issue but when setting an animation to be triggered (pinwheel when holding space bar) seems to up CPU usage dramatically. When animation disabled CPU usage drops again.

ccMSC, 2015-09-08:
The "show animated" checkbox does increase CPU usage, mostly because it has to constantly update the window. There's not much that can be done about that. See discussion in #67.

closed 2015-09-08 by ccMSC

securify, 2016-04-15:
Same issue here.  El Capitan on a new retina 13inch macbook pro.  CPU usage seems to jump while connected with spikes up to 45% on single core and when disconnected it fires more often  and is consistent about 15%.