ccMSC/ckb #121: GUI hangs on color-chooser close

bkloppenborg, 2015-09-09:
Whenever I try to pick a key color the `ckb` GUI hangs after I click the "OK" button in the color chooser. If I select a profile the keys all change correctly.

## Details:

K70 RGB (Firmare 1.30) on Ubuntu Linux 15.04 using latest git master version (2eed29a4f60bfb38085f152663f60f397fc2edd7) of ckb

## Steps to reproduce:

1. I removed my old `ckb` configuration directory located in `~/.config/ckb`
2. Launch `ckb`
3. Choose any profile (e.g. `HWProfile1` or `ckb Demo` or make a new profile)
4. Select any key, click change color
5. Select any color (or just leave it as the color originally selected)
6. Click "Ok"
7. `ckb` hangs indefinitely.

(As a reminder, I had a similarly odd issue with #85 which was resolved by deleting my ckb configuration directory. I tried that for this issue as well and it didn't fix the problem)

ccMSC, 2015-09-09:
Strange, not seeing it on my machine. Can you do a `gdb` backtrace from where the freeze occurs?

bkloppenborg, 2015-09-13:
After doing some debugging today I tracked the problem down to [colorbutton.cpp:61](https://github.com/ccMSC/ckb/blob/master/src/ckb/colorbutton.cpp#L61) where the `QColorDialog` modal dialog is created. Replacing that line with a hard-coded color (like `QColor newColor = QColor(Qt::blue);`) makes `ckb` behave as it should.

There was a [similar bug in QT 4.6.0](https://bugreports.qt.io/browse/QTBUG-5733) but appears to have been fixed. My version of QT (5.4.1 installed via. apt-get on Ubuntu 15.04) always hangs. After upgrading QT to version 5.5 via. [QT's download page](http://www.qt.io/download-open-source/) the color chooser works as expected.

Thus this appears to be a bug in QColorChooser in QT 5.4.1 rather than an issue in `ckb`. Feel free to close the issue if you do not with for me to investigate further.

ccMSC, 2015-09-15:
Alright, sounds like there's nothing I can do about it. I'll consider advising people to download Qt manually if their system version isn't up to date.

closed 2015-09-15 by ccMSC