ccMSC/ckb #122: Latest Binary, k95 not working on boot/reboot

Dugthefreshest, 2015-09-10:
Good day all. 

I got my k95 today and set it up via the instructions here, all in good order.   I saw some other threads about this same issue, but I couldn't solve my problem.  I reinstalled and tried to use the "quickinstall"  but wasn't having any luck getting that to work.  I got xtools and qt5 but was getting a message saying I didn't have qt5 when doing the quick install. 

If you need any reports, just let me know how to generate them.   Im in love with the kb, just having issues.  Works fine through boot camp.  I got it to work again after an unplug, re plug.  Also at the log in i could hit the switch in the back, but ckb did not recognize it afterwards. 

Thanks in advance. This is a great project and great community.

Dugthefreshest, 2015-09-11:
When trying to run the quick install, i get this 
Preparing build files...
No suitable qmake found.
Please make sure you have Qt5 installed.
Re-run this program with QMAKE=/path/to/qmake to select qmake manually.

ccMSC, 2015-09-11:
Please follow the instructions from this post for generating a log file: https://github.com/ccMSC/ckb/issues/106#issuecomment-131287945
(you can do this with the binary release, no need to compile it)

This should provide some info about what it's getting stuck on.

Dugthefreshest, 2015-09-12:
alrighty, so after a while this is how far I got.  I did both sudo commands, and on reboot, the keyboard works ,but not getting any light patterns or anything.  ckb shows "driver inactive".I suppose this is why my keyboard is actually working like a keyboard, but not giving me the light show.

edit
I decided to try this after the reboot :
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
Password:
    ckb: Corsair RGB driver beta-v0.2.1
[I] Root controller ready at /var/run/ckb0
[I] Connecting  (S/N: )
[I] Starting input thread for /var/run/ckb1
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:15): Got return value 0xe0004051
[E] _usbrecv (via firmware.c:15): Too many send/recv failures. Dropping.
[W] start_dev (device.c:24): Unable to load firmware version/poll rate
[I] corsair k95 ready at /var/run/ckb1

After I did this the lights are working again.

  I edited the plist and rebooted, but the logs are not in the location specified. I did this entire process a few times to make sure, and I am not getting anything at all unfortunately.  

Any ideas? Sorry for being a newb, im pretty computer savvy, but this is a bit out of my league as for as investigating issues.

ccMSC, 2015-09-15:
Possibly the log files are not displaying in Finder? You can check by running the Terminal command:
`ls /var/log | grep ckb`

If you see them there, you can get the output with
`cat /var/log/ckb.log`
and `cat /var/log/ckb.err.log`

Otherwise, double-check the daemon plist to make sure it's been replaced correctly and that it matches the version I posted exactly (nothing missing at the beginning or end).

closed 2016-01-15 by ccMSC