ccMSC/ckb #123: CKB High CPU Usage and Unable to Detect Keyboard

linjer, 2015-09-10:
This is related to my previous issue filed #109 .

I installed the latest release of CKB and it's been mostly error free. But I also didn't get the chance to setup logging. As an aside, can you please enable logging by default, as the log files appear to be very small.

Different from the previous issue, the ckb-daemon CPU usage seems to be more around 50%. 
[image]

I'm typing this issue using the keyboard, but the CKB GUI says no keyboard detected. Lighting profiles are not set.

Environment:
OSX 10.10.3
K75 RGB

linjer, 2015-09-10:
Killing the `ckb-daemon` process then restarting the CKB GUI does resume normal use.

Dugthefreshest, 2015-09-11:
Killing that process definitely worked.

Dugthefreshest, 2015-09-11:
It is note though, that after rebooting, had the same issue. My current mac version is 10.10.5

ccMSC, 2015-09-11:
Fairly certain this is the same issue as #122.

closed 2015-09-11 by ccMSC

linjer, 2015-09-11:
I will be upgrading to 10.11 soon, which may bring a whole other world of issues. :P