ccMSC/ckb #124: Keyboard hangup

master-caster, 2015-09-13:
Hallo and nice job creating this software.

My problem is a generic problem and doesn't base on the code here.
But after researching i hope to get the right hint here.

Using the k95 in bios mode won't "break" the keyboard after pressing the G-keys or anything
other special except the media keys.

But if i have the keyboard switched to mode "1" or "2"(anything other than bios
 (and it works only on reboot)) i can use the
keyboard as long as i don't use the G-keys or the M1, M2 and M3 key. After pressing
one of these keys the keyboard becomes fully useless.

Using dmesg(with another keyboard plugged in ;-) ) won't show any issues the keyboard just stops working. To get it back online i have to disconnect it physically and reconnect.

I have updated the keyboard to the latest firmware using a windows client.
Any idea how to fix or at least workaround this behaviour?
Anyone having the same problem?

ccMSC, 2015-09-15:
I assume you're not running `ckb-daemon`? I'd recommend installing it if possible, as it will fix this issue. You don't have to use the GUI if you don't want RGB control as the driver works just fine without it. You might also be able to fix the issue by adding `cmdline` parameters to your kernel, as outlined in the Troubleshooting section.

master-caster, 2015-09-15:
Thank you for your response. I installed the daemon and now the keyboard won't hangup.

I think i can keep it that way for now. So this issue is fixed closed ;-)

closed 2015-09-15 by master-caster