ccMSC/ckb #125: Parameters for ckb Program

Keridos, 2015-09-14:
Could you maybe add in documentation about the parameters of the ckb program? If there are none, could you please add in a way to start it in tray mode since I am currently trying to get the program started via my xinitrc/xsession script.

ccMSC, 2015-09-15:
There are currently three:
* `--version` prints the version of ckb and then immediately exits. This can also be achieved with `cat /dev/input/ckb0/version`
* `--background` launches silently without displaying the window. This is the default behavior when starting at login.
* `--close` does not launch the program but instead causes the already-running instance (if any) to exit.

TBH, I don't really consider them useful enough to document

Keridos, 2015-09-15:
You could just make a "--help" print this out.