ccMSC/ckb #129: K40 Keyboard support

debianmain1, 2015-09-27:
Greetings!!!  As per request, I'm opening this & will dual post here & on the Corsair forms (unless you think that would be spamming the forum thread).

First information after installing clement's K90 driver with the K40 mod:

lsusb reports: Bus 003 Device 008: ID 1b1c:1b0e Corsair 

for f in /sys/bus/hid/devices/0003\:1B1C\:*/report_descriptor; do echo $f; xxd -g 1 "$f"; done
/sys/bus/hid/devices/0003:1B1C:1B0E.001B/report_descriptor
00000000: 05 01 09 06 a1 01 05 07 19 e0 29 e7 15 00 25 01 ..........)...%.
00000010: 75 01 95 08 81 02 75 08 95 01 81 01 05 07 19 00 u.....u.........
00000020: 2a ff 00 15 00 26 ff 00 75 08 95 06 81 00 05 08 *....&..u.......
00000030: 19 01 29 03 25 01 75 01 95 03 91 02 95 05 91 01 ..).%.u.........
00000040: c0 .
/sys/bus/hid/devices/0003:1B1C:1B0E.001C/report_descriptor
00000000: 05 0c 09 01 a1 01 05 0c 19 00 2a ff 0f 15 00 26 ..........*....&
00000010: ff 0f 75 10 95 02 81 00 c0 ..u......
/sys/bus/hid/devices/0003:1B1C:1B0E.001D/report_descriptor
00000000: 05 01 09 06 a1 01 05 07 19 e0 29 e7 15 00 25 01 ..........)...%.
00000010: 75 01 95 08 81 02 19 00 29 6f 15 00 25 01 75 01 u.......)o..%.u.
00000020: 95 70 81 02 c0

cvuchener, 2015-09-27:
Does the keyboard have per-key led settings like other RGB keyboard or is it only a global setting (all the keys at once but per-profile)?

debianmain1, 2015-09-27:
Does the keyboard have per-key led settings like other RGB keyboard or is it only a global setting (all the keys at once but per-profile)? 

Global setting seems to be all it has (per "M" key 1~3) Your edit is correct--

debianmain1, 2015-09-27:
First software sniffer run.
In order:
Creat new profile2
Set M1 color
Set M2 Color
Set M3 color
Upload to keyboard
Assign G1 through G6 keys in M3
Upload to keyboard
Change back profile to profile1
close software

Seem to be running into not being able to rename the capture file & uploading it here--I will upload it to the forum & link here:  http://forum.corsair.com/forums/showthread.php?p=810811#post810811

cvuchener, 2015-09-27:
It would have been better to have one capture per action. But actually most of the request are the same as for the K90, so I understand most of it. Knowing the exact colors you used would very helpful though.

The profile writing looks exactly like the K90 but I did not see any macro or key assignment. Are you sure you sent them to the hardware. Profiles are written three times during the capture, all are empty.

The only new request here is 51. It is sent three times (on for each profile I guess) when the profile are sent to the hardware. It is also sent when switching to a profile either by pressing the key or from the software (what you called "Creat new profile2"). There more of these request I cannot explain.

cvuchener, 2015-09-27:
The high byte of wIndex could mean 0 for temporary color change, or 1 to 3 for changing a profile color. That leaves three bytes (low byte from wIndex and two bytes from wValue) that could be the color.

cvuchener, 2015-09-27:
Can you try [this code](https://github.com/cvuchener/k40-set-color)? You need the development package for libusb-1.0 to compile it.
Test `./k40-set-color 255 0 0`, `./k40-set-color 0 255 0` or `./k40-set-color 0 0 255`. What happens?

debianmain1, 2015-09-27:
Sorry I didn't include colors--do you want another set of runs with descrete actions for each?
The "Create new profile 2" is in the profile screen & I just selected "new profile" from the list.
This was a unlocked profile that I could edit--

debianmain1, 2015-09-27:
I'll check out your code in a couple of minutes...

debianmain1, 2015-09-27:
I had the .dev lib for libusb installed already--cd'd into the set-color folder & did a make...

dean@linux:~/Desktop/k40-set-color-master$ make
g++ -M -Wall -std=c++11 -g -O0 -I/usr/include/libusb-1.0   main.cpp > main.deps
g++ -c main.cpp -Wall -std=c++11 -g -O0 -I/usr/include/libusb-1.0   -o main.o
g++ main.o -lusb-1.0   -o k40-set-color

Looks good, but when I execute--nothing happens 
dean@linux:~/Desktop/k40-set-color-master$ sudo ./k40-set-color 0 255 0  = no change

cvuchener, 2015-09-27:
> Sorry I didn't include colors--do you want another set of runs with descrete actions for each?

It is okay for me (maybe MSC will want more). But we need the exact color for each setting (give the numbers for red, green and blue).

> "Create new profile 2"

You cannot create or delete (hardware) profiles. You have three, one for each button. Clicking this button switched the current profile, that's all.

> Looks good, but when I execute--nothing happens 

No luck for the first guess (or do I have a bug in the program?). I saw a small error for the "red" channel, but it would not have affected the other test.

cvuchener, 2015-09-29:
Please update k40-set-color.

Test setting a profile color (use options `-p1` or `-p2` or `-p3` and try different colors).

Then test setting the current color after some initialization request (from the first capture you provided). In the Makefile, there are two lines:

    #CXXFLAGS+=-DREQUEST_56
    #CXXFLAGS+=-DREQUEST_50

Try uncommenting one at a time, then both. For each test, recompile (`make -B`) and try setting the current color (without the `-p` option).

ccMSC, 2015-09-29:
For starting out, I'm only really interested in the RGB commands. If it's like the K70, the hardware saving capabilities are probably pretty limited. Could you give me some captures for setting specific keys (i.e. A, S, D, etc...)?

debianmain1, 2015-09-29:
Sounds good--I'll get both runs this evening--about 7pm pacific

debianmain1, 2015-09-30:
Run without flags set.....

dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 255 0 0
dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 0 255 0
dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 0 0 255
dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 255 255 255
dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 255 255 100

All run the color(s) I expected blue, green, red, white, very light blue.

uncommenting CXXFLAGS+=-DREQUEST_56 locks up the application
uncommenting CXXFLAGS+=-DREQUEST_50 gives a status after request line.

dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 175 225 100
Status after request 50: 01 03 06 00 64 e1 af 03 00 01
dean@linux:~/Desktop/k40-set-color-master$
sudo '/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 0 0 255
Status after request 50: 01 03 06 00 64 e1 af 03 00 01
dean@linux:~/Desktop/k40-set-color-master$
sudo '/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 0 255 0
Status after request 50: 01 03 06 00 ff 00 00 03 00 01
dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 255 255 255
Status after request 50: 01 03 06 00 00 ff 00 03 00 01
dean@linux:~/Desktop/k40-set-color-master$ sudo
'/home/dean/Desktop/k40-set-color-master/k40-set-color' -p3 255 0 0 Status
after request 50: 01 03 06 00 ff ff ff 03 00 01

uncommenting both locks up the application.

Looks like DREQUEST_50 will give back possibly useful information.

After 3 or so color changes it will also lock up the app--force quitting
the terminal & restarting it will allow several more color changes until it
happens again..needs a "reset" ?

debianmain1, 2015-09-30:
According to the website--it only has 36K worth of "storage", so not much
is right.....

OK, I'm going to set P1 with G1/W G2/A G3/S G4/D G5/O G5/M then do the same
for P2 & P3. will do a run for each profile.

Hope these are what you are looking for...

-- 
Cheers!!!
Dean Loros
Performance by Design Ltd.
autocrosser at http://forums.linuxmint.com

ÔÃ²¡          ÿÿ  ù   Ù<Vøf *   *     <‰ÿÿÿÿ    	   ƒ                  Ù<Vþ£ *   *     9<‰ÿÿÿÿ    	   ƒ                  â<V× *   *     <‰ÿÿÿÿ    	   ƒ                 â<VSv *   *     9<‰ÿÿÿÿ    	   ƒ                  â<VY³ *   *     <‰ÿÿÿÿ    	   ƒ                  è<VdÕ *   *     9<‰ÿÿÿÿ    	   ƒ                 è<VŸ7 *   *     <‰ÿÿÿÿ    	   ƒ                  è<V¥t *   *     9<‰ÿÿÿÿ    	   ƒ                  ï<Vù *   *     <‰ÿÿÿÿ    	   ƒ      @           ï<V4ò *   *     9<‰ÿÿÿÿ    	   ƒ                  ï<V:/ *   *     <‰ÿÿÿÿ    	   ƒ                  õ<V!ã	 *   *     9<‰ÿÿÿÿ    	   ƒ    €             õ<V€³ *   *     <‰ÿÿÿÿ    	   ƒ                  õ<V†ð *   *     9<‰ÿÿÿÿ    	   ƒ                  ü<V¸ *   *     <‰ÿÿÿÿ    	   ƒ                 ü<V´Ý	 *   *     9<‰ÿÿÿÿ    	   ƒ                  ü<Vº
 *   *     <‰ÿÿÿÿ    	   ƒ                  =Vy *   *     9<‰ÿÿÿÿ    	   ƒ                 =VPÛ *   *     <‰ÿÿÿÿ    	   ƒ                  =VV *   *     9<‰ÿÿÿÿ    	   ƒ                  	=Vîì $   $    pNŠÿÿÿÿ             @     	=Vîì        pNŠÿÿÿÿ            
=V¸ $   $    ¸ú®‰ÿÿÿÿ        €    À     
=V¸        ¸ú®‰ÿÿÿÿ       €   
=V¸        ¸ú®‰ÿÿÿÿ       €    
=V¸ $   $    ¸ú®‰ÿÿÿÿ             @   % 
=V¸ A   A    ¸ú®‰ÿÿÿÿ        %    _                                  
=V¸        ¸ú®‰ÿÿÿÿ            
=VSW $   $    %=‰ÿÿÿÿ        €    À     
=VSW        %=‰ÿÿÿÿ       €   
=VSW        %=‰ÿÿÿÿ       €    
=V| $   $    ðê¹‰ÿÿÿÿ        €    À     
=V|        ðê¹‰ÿÿÿÿ       €   
=V|        ðê¹‰ÿÿÿÿ       €    
=V| $   $    %=‰ÿÿÿÿ             @   % 
=V| A   A    %=‰ÿÿÿÿ        %   ÐÑÒÓÔÕ                        
=V|        %=‰ÿÿÿÿ            
=V½¡ $   $     :‰ÿÿÿÿ        €    À     
=V½¡         :‰ÿÿÿÿ       €   
=V½¡         :‰ÿÿÿÿ       €    
=V½¡ $   $     :‰ÿÿÿÿ             @3  Y  
=V½¡         :‰ÿÿÿÿ            
=VÕ•	 $   $    ˆø®‰ÿÿÿÿ             @   % 
=VÕ•	 A   A    ˆø®‰ÿÿÿÿ        %    _                                  
=VÕ•	        ˆø®‰ÿÿÿÿ            
=V5 $   $    à“:‰ÿÿÿÿ        €    À     
=V5        à“:‰ÿÿÿÿ       €   
=V5        à“:‰ÿÿÿÿ       €    
=V?à $   $    K²‰ÿÿÿÿ        €    À     
=V?à        K²‰ÿÿÿÿ       €   
=V?à        K²‰ÿÿÿÿ       €    
=V?à $   $    ÈwÑ‰ÿÿÿÿ             @   % 
=V?à A   A    ÈwÑ‰ÿÿÿÿ        %   ÐÑÒÓÔÕ                        
=V?à        ÈwÑ‰ÿÿÿÿ            =V@= $   $    ØÊ	Šÿÿÿÿ        €    À     =V@=        ØÊ	Šÿÿÿÿ       €   =V@=        ØÊ	Šÿÿÿÿ       €    =V@= $   $    ØÊ	Šÿÿÿÿ             @3$  =V@=        ØÊ	Šÿÿÿÿ            =VX1 $   $    `èÄ‰ÿÿÿÿ             @   % =VX1 A   A    `èÄ‰ÿÿÿÿ        %    _                                  =VX1        `èÄ‰ÿÿÿÿ            =V™Ð $   $    ˆÑŠÿÿÿÿ        €    À     =V™Ð        ˆÑŠÿÿÿÿ       €   =V™Ð        ˆÑŠÿÿÿÿ       €    =VÂ{ $   $    ðê¹‰ÿÿÿÿ        €    À     =VÂ{        ðê¹‰ÿÿÿÿ       €   =VÂ{        ðê¹‰ÿÿÿÿ       €    =VÂ{ $   $    0¯‰ÿÿÿÿ             @   % =VÂ{ A   A    0¯‰ÿÿÿÿ        %   ÐÑÒÓÔÕ                        =VÂ{        0¯‰ÿÿÿÿ            =V	 $   $    èŠÿÿÿÿ        €    À     =V	        èŠÿÿÿÿ       €   =V	        èŠÿÿÿÿ       €    =V	 $   $    èŠÿÿÿÿ             @3l«Y  =V	        èŠÿÿÿÿ            =V
 $   $    ˆÑŠÿÿÿÿ             @0     =V
        ˆÑŠÿÿÿÿ            =V4Î	 *   *     <‰ÿÿÿÿ    	   ƒ                 =Vjó *   *     9<‰ÿÿÿÿ    	   ƒ                  =Vo0 *   *     <‰ÿÿÿÿ    	   ƒ                  =V–	 *   *     9<‰ÿÿÿÿ    	   ƒ                 
ÔÃ²¡          ÿÿ  ù   *=VÁ  *   *     9<‰ÿÿÿÿ    	   ƒ                  *=VÇÝ *   *     <‰ÿÿÿÿ    	   ƒ                  /=VTä  $   $    ¨–¯‰ÿÿÿÿ             @     /=VTä         ¨–¯‰ÿÿÿÿ            /=Ve› #   #    pôú‰ÿÿÿÿ    	              /=Vq $   $    @”¯‰ÿÿÿÿ             @3$   /=Vq        @”¯‰ÿÿÿÿ            3=V–ñ *   *     9<‰ÿÿÿÿ    	   ƒ                 4=V£È *   *     <‰ÿÿÿÿ    	   ƒ                  4=V© *   *     9<‰ÿÿÿÿ    	   ƒ                  :=V/) *   *     <‰ÿÿÿÿ    	   ƒ                 ;=V$ *   *     9<‰ÿÿÿÿ    	   ƒ                  ;=V*I *   *     <‰ÿÿÿÿ    	   ƒ                  @=V‹ä *   *     9<‰ÿÿÿÿ    	   ƒ [... cut ...]    pî¯‰ÿÿÿÿ       €   ¨=V8p        pî¯‰ÿÿÿÿ       €    ¨=V8p $   $    pî¯‰ÿÿÿÿ             @3l«Y  ¨=V8p        pî¯‰ÿÿÿÿ            ¨=VPd $   $    pþ‰ÿÿÿÿ             @0     ¨=VPd        pþ‰ÿÿÿÿ            ­=V¢  *   *     <‰ÿÿÿÿ    	   ƒ                 ­=VÝj *   *     9<‰ÿÿÿÿ    	   ƒ                  ­=Vâ§ *   *     <‰ÿÿÿÿ    	   ƒ                  ®=Vzw *   *     9<‰ÿÿÿÿ    	   ƒ                 

debianmain1, 2015-09-30:
As I use the software in windows, it seems that Corsair is a bit optimistic
about how this keyboard changes color--not on a per key basis--just per
profile globally on all keys. I have not tried any animations in Windows
(not really a windows user), but I will explore the limits of the software
to see what is "really" possible. I do know it will record & save macros
per profile (have seen some output on this-if you want a capture of this
I'll get it done) & will report back as to what else it can do.

debianmain1, 2015-09-30:
OK--what it can do is pulse the preset color (vary brightness dim to max
with a variable time setting) or do a variable color change through the
spectrum, also with a variable time delay--there is also an option to do
this only after Idle (no setting for how long idle time is-looks like 30
sec of non-useage) or on all the time. I set it to P2 to see if it needs
the windows driver to do this & it looks like it does.

So, rather more basic than it's more expensive brothers, but I can see that
some interesting things could be done with scripts.

debianmain1, 2015-10-01:
What do you guys want checked next?

ccMSC, 2015-10-03:
Hmm...if that's the case I may not have the ability to add it in ckb, honestly. The driver is meant to handle per-key RGB animations and won't really do anything else without a huge rewrite. I think you're better off helping Clement get it into a kernel driver since it seems like the keyboard has more in common with the K90 anyway.

debianmain1, 2015-10-03:
OK--its looking like the K90 driver is working well for this keyboard & Clement has got a good percentage of its capabilities figured out. Will keep posting here if you are interested......

closed 2016-01-15 by ccMSC