ccMSC/ckb #130: Does Sabre work at all/planned to work for ckb?

evilarceus, 2015-09-29:
I am an owner of a Corsair Sabre and I was really stumped that I couldn't change the lighting effects whatsoever on Ubuntu. Is it planned at all? I understand how long it would take to develop this, but is it at least planned?

ccMSC, 2015-09-29:
I've begun testing for it - if you could help out, that would be much appreciated. There's a branch for it here: https://github.com/ccMSC/ckb/tree/sabretest

This should connect the mouse and treat it as an M65. Could you try this version and tell me which lighting zones work (if any)?

evilarceus, 2015-10-01:
Alright, I'll try it and see how it goes! Thanks!

evilarceus, 2015-10-02:
Hmm, doesn't appear at all when I plug it in. I did a full reinstall too.

ccMSC, 2015-10-03:
Hmm...possible I added the wrong vendor/product by mistake? Can you give me the output of `lsusb -d 1b1c:`?

It would also be helpful to get a log from the daemon. To see the output, first stop the system service (`sudo systemctl stop ckb-daemon` for systemd, or `sudo service ckb-daemon stop` for Upstart) and then run `sudo /usr/bin/ckb-daemon` in a terminal. If it's picking up the device it should print something like "Connecting Corsair Sabre RGB".

adagari, 2015-11-30:
Hey,
Not sure if you received the info you were looking for with your last post. I installed ckb from the testing branch and like above wasn't detected by my distro (mint). I ran the command "lsusb -d 1b1c:" and received output "Bus 002 Device 004: ID 1b1c:1b14 Corsair". I also added "usbhid.quirks=0x1B1C:0x1B14:0x408" to the kernel boot parameters, but i'm not sure if this is right as it did not show any difference in the application. Lastly, I execute your instuctions to display output in the terminal as above and received 

"ckb: Corsair RGB driver beta-v0.2.2+t02
[I] Root controller ready at /dev/input/ckb" 

for the output. If there is anything else I can help you do please let me know.

ewhac, 2015-12-03:
Well, I *was* going to submit a patch for the SABRE Optical (ProductID 0x1b14), but then I noticed in the Git history that you disabled it without explanation.

Anyhoo, after adding my own support code on the master branch and recompiling, the SABRE was recognized by the daemon, and ckb picked it up and displayed a UI for it.  I'm not sure what I should have expected, but only two of the four LEDs were set to 'wave' mode ('logo' and 'front light').  The other two were solid.

What sorts of things do you need tested?

ccMSC, 2015-12-03:
Interesting that you mention `1b14`, since the product ID that I had was `1b19`. I'd guess that 14 is optical and 19 is laser. So I can just update the code to treat them as the same device.

I probably never updated the demo profile for anything other than the M65. Do all four lights work if you edit them? The DPI light should be controllable from the performance tab, the others can be assigned in the lighting tab.

ewhac, 2015-12-05:
> Do all four lights work if you edit them?

Yes.

> The DPI light should be controllable from the performance tab, [ ... ]

It is.

ccMSC, 2015-12-05:
Excellent! I've updated the testing repo so it now handles both the laser and optical versions. They should behave identically. @adagari can you try it out and see if it's working now?

adagari, 2015-12-06:
Hey ccMSC,
Happy to report that all seems to be functioning with the sabre under the new testing build. Only bug I have noticed so far is under the lighting control you have the option to change the LED color for the DPI switch. That option should be reserved for the different DPI levels under the performance tab. Under the stage you have the option to set the color. Basically, the option to change the color should be removed from the lighting tab. If anything new is noticed I will be sure to report.

ewhac, 2015-12-06:
> ...you have the option to change the LED color for the DPI switch. That option should be reserved for the different DPI levels under the performance tab.

Dissent.  I think it should remain a user-selectable option.  It should *default* to the DPI settings, but the user should be able to change it to wibble along with all the other LEDs.

ewhac, 2015-12-06:
> I've updated the testing repo so it now handles both the laser and optical versions.

Your changes are effectively identical to the ones I made.  Awesome!

adagari, 2015-12-06:
It is user-selectable as in you can change the color for any selected DPI, but if you change the color from the lighting control tab, it causes it to flicker back and forth.

ccMSC, 2015-12-06:
I'm not experiencing any flickering when the DPI indicator is enabled. Are you seeing it in the GUI, or on the physical mouse, or both?

The design here, and I'll probably add some text to clarify this later, is that you can still assign colors/animations to that particular light, but the indicator overrides them. If you disable the indicator, you can see them again. There's also an option to adjust the opacity of the indicator, so that you can partially see the lighting/animations, but still have the color influenced by the indicator light.

adagari, 2015-12-06:
Scratch what I said before, it seems to be functioning as intended. You can change the color in the GUI but the DPI color is still shown. I think I might of mistaken the flickering for something else...

closed 2016-01-15 by ccMSC