ccMSC/ckb #131: Can't compile newfx branch

KraXarN, 2015-10-01:
I wanted to try to compile the newfx branch to try out the new effects, but get the following error while trying to compile it

main.o: In function `ckb_init':
/home/kraxarn/Downloads/ckb-newfx/src/ckb-mviz/main.c:51: undefined reference to `pa_simple_new'
main.o: In function `getFreqDec':
/home/kraxarn/Downloads/ckb-newfx/src/ckb-mviz/main.c:93: undefined reference to `pa_simple_read'
collect2: error: ld returned 1 exit status
make[1]: *** [../../bin/ckb-animations/ckb-mviz] Error 1
make: *** [sub-src-ckb-mviz-make_first] Error 2
make: *** Waiting for unfinished jobs....

It seems something is wrong with the required libpulse-simple, which is replaced by libpulse-dev in Ubuntu 15.04, which is installed and updated. I also just tried reinstalling the package, but it still gives off the same error.
I also tried manually going into the ckb-mviz folder and tried to jump compile that without luck.

tatokis, 2015-10-01:
From a quick look I had, if you edit the generated makefile inside ckb/src/ckb-mviz/ and change line 175 to
```
$(LINK)  -o $(TARGET) $(OBJECTS) $(OBJCOMP) $(LIBS) $(LFLAGS)
```
and then run make manually on that directory it seems to compile without any issues.
It calls g++ with the -lpulse-simple argument after the .o files
```
g++  -o ../../bin/ckb-animations/ckb-mviz main.o kiss_fft.o kiss_fftr.o     -m64 -lpulse-simple
```

ccMSC, 2015-10-03:
You're probably missing the pulseaudio development libraries. It's meant to be an optional dependency (won't be used on OSX at all), which is why I put it into a different branch for now.

I've updated the branch to something that should allow the project to compile without Pulse. The visualizer should simply be skipped if the library isn't available.

tatokis, 2015-10-03:
I am also on 15.04, and I do have the library installed as I can compile my own application and link against it.
I don't think the problem is that the library does not exist.

ccMSC, 2015-10-03:
@ruler501 any idea what's going on here? It compiles fine on my system (Arch), seems really odd that it would fail to link if the library's installed correctly.

I would have assumed that there was some package missing, but if you were able to compile a different application with it then that seems unlikely.

tatokis, 2015-10-03:
@ccMSC Have a look at this
https://vps.tasossah.com/misc/ckb-libpulse.webm

JackFarrand, 2015-11-22:
it compiles and runs cleanly on fedora 23, but doesn't respond to any playing music, using either the power or magnitude presets.  Any ideas on how to debug?

KraXarN, 2015-11-23:
I have since then switched to OpenSUSE where it also compiles just fine.

I later opened up pavucontrol and there was a new entry under "Recording". I changed that to a monitor of my sound card and it now works fine :D

JackFarrand, 2015-11-24:
Thanks for the info!

Aye, tuning / changing the gradient for the function should fix the white-out.  Perhaps I should commit a 'how to' section to this branch's Readme.md for others to get this working.

SicVolo, 2016-01-04:
@JackFarrand Could you post a comment here on how you got it working?

JackFarrand, 2016-01-05:
@SicVolo as KraXarN said, run pavucontrol in a terminal and look for the recording device and change the input to a monitor of your chosen sound output, the keyboard then responds properly.  The white-out issue can be fixed by changing the blending settings of the music effect.