ccMSC/ckb #133: bind key to execute command inconsistent

znorris, 2015-10-05:
Keyboard: K70 (solid red)
Firmware: 1.09
ckb: v0.2.1
OS: Ubuntu 15.04
3.19.0-30-generic #33-Ubuntu SMP Mon Sep 21 20:58:04 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

When attempting to bind a key to 'launch program on keypress' I receive inconsistent results. If I set the program to the given example, `xdg-open /home/user` then it executes the command without issue. However, when I set the command to anything else... `firefox`, `gnome-terminal`, `xterm`. They do not open.

I recently wiped and upgraded from Ubuntu 14.04 to 15.04. I'm not sure what version of ckb I was using before but binding programs to keys was no problem before. Is there a ckb error log I can go check?

tatokis, 2015-10-05:
I can reproduce this issue on Ubuntu 15.04.
It appears it is related to the Xorg DISPLAY environment variable.

For example a workaround to start firefox or thunderbird from within ckb is to do either
```
env DISPLAY=:0 /usr/bin/thunderbird
or
/usr/bin/thunderbird --display=:0
```
Making a little script that redirects stderr and stdout from thunderbird to a file, and then running it via ckb reveals that the display is indeed somehow the issue.
```
thunderbird.sh
#!/bin/bash
/usr/bin/thunderbird &> ~/displaytest.txt

displaytest.txt:
(process:22702): GLib-CRITICAL **: g_slice_set_config: assertion 'sys_page_size == 0' failed
Error: cannot open display: :0.
```

ccMSC, 2015-10-07:
Try the latest version from the testing repository - it should have this issue fixed.

tatokis, 2015-10-07:
I tried the latest version from testing, but nothing changed.

I decided to dump the environment variables ckb sets on the application it launches and I noticed that DISPLAY was being set to ":0." instead of ":0"

I don't have any non-Ubuntu machines to test this, so I didn't make a pull request, but the fix for me was to remove the last character from "envstr" in src/ckb/keyaction.cpp.

```
diff --git a/src/ckb/keyaction.cpp b/src/ckb/keyaction.cpp
index 3973d2b..6923316 100644
--- a/src/ckb/keyaction.cpp
+++ b/src/ckb/keyaction.cpp
@@ -415,7 +415,7 @@ void KeyAction::adjustDisplay(){
         *ptr = '\0';
 
     snprintf(buf, sizeof(buf), ".%i", XScreenNumberOfScreen(attr.screen));
-    strncat(envstr, buf, envstr_size - 1 - strlen(envstr));
+    strncat(envstr, buf, envstr_size - 2 - strlen(envstr));
     setenv("DISPLAY", envstr, 1);
 
     delete[] envstr;
```
If this breaks the feature on other distros/setups then a possible solution would be to check if the string ends with a "." character and remove it.

ccMSC, 2015-10-09:
It's actually supposed to say `:0.0` - looks like the lack of a dot in the original string was throwing it. As of bb5ec36246fc82af2ec7fd034cb21004d543d9d0 this should be handled correctly now.

If it's still not working for some reason, I'll have to add an option to disable this feature.

tatokis, 2015-10-10:
@ccMSC Fair enough, didn't know about the dot.
Either way that commit fixed it for me. Thanks a lot!

olofson, 2015-10-29:
Got the same problem after upgrading from v0.1.something to v0.2.1 on Ubuntu Studio, but it works fine again with v0.2.1+t12.

closed 2015-11-11 by ccMSC