ccMSC/ckb #136: All media keys scept volume doesn't work?

KraXarN, 2015-10-07:
Hello!
I've recently had this issue where the all media keys except the volume scroll wheel and the mute button not working, including the macro keys.
Even if I try to bind them to different keys, it still won't work at all.
Thanks.

KB: K95 RGB
OS: Kubuntu 15.04 x64

ccMSC, 2015-10-09:
Is the keyboard in BIOS mode (blinking scroll lock light)? They don't work in BIOS mode due to a hardware limitation.

If not, do they work if you close the GUI?

KraXarN, 2015-10-09:
I'm not really sure about that really, the switch is set to BIOS mode 2, which is what I need to make it detectable by BIOS, but it has also always worked.
But there is no scroll lock light blinking.
I could try and see if they work under Windows when I boot into it next time.

And my bad for not checking without the ckb driver loaded, but it seems to still not work, so maybe it's something with the keyboard itself rather than the driver...

ccMSC, 2015-10-14:
huh. I don't recall making any changes related to this, but maybe there was a bug I sorted out by accident :P Glad to hear it's working, anyway.

An update notifier would be a nice feature to have, I agree. Maybe I'll try to get that in with v0.3 or something.

closed 2015-10-14 by KraXarN