ccMSC/ckb #138: Tried to compile Strafe RGB

rinkelj, 2015-10-10:
1 - Had to convert to machos 10.11
2 - undeclared identified P_STRAFE_RGB
Verified that STRAFE_RGB isn't defined in usb.h

Unable to make

rinkelj, 2015-10-10:
Sorry macosx 10.11

rinkelj, 2015-10-10:
debug output:
aemon/usb_mac.c
../../../ckb/src/ckb-daemon/usb_mac.c:399:76: error: use of undeclared identifier 'P_STRAFE_RGB'
    int products[] = { P_K65, P_K70, P_K70_NRGB, P_K95, P_K95_NRGB, P_M65, P_STRAFE_RGB };
                                                                           ^
../../../ckb/src/ckb-daemon/usb_mac.c:406:31: error: invalid application of 'sizeof' to an incomplete type 'int []'
    for(uint i = 0; i < sizeof(products) / sizeof(int); i++){
                              ^~~~~~~~~~
2 errors generated.

closed 2015-10-11 by rinkelj

SicVolo, 2015-10-12:
Change P_STRAFE_RGB to P_STRAFE. My bad, I've not tried compiling it on mac.

SicVolo, 2015-10-13:
@rinkelj I assume it compiled, since you closed the issue. Care to share the other changes you did to make it work on OS 10.11. This way we can change it for everybody.

rinkelj, 2015-10-14:
I logged a new issue with a complete explanation for others to follow.

It compiles and runs nicely.
Please see issue 141 with the complete description for reference.

Works good so far..