ccMSC/ckb #139: On OSX 10.10.5 restart, need to plug-unplug data cable for ckb to recognize K65 RGB

notbrain, 2015-10-10:
Every time I reboot, the K65 goes into a default mode and is not recognized at all (keys do not register), and is always fixed by me simply re-plugging in the data side USB cable into the back of my non-Thunderbolt Cinema display.

When I launch ckb in this state, it simply reports "no devices connected."

Is there anything I can provide to help troubleshoot this? Anyone else have this issue?

rbf192, 2015-10-11:
Happens to me too

SicVolo, 2015-10-12:
Have you tried the usb quircks from the readme? That fixed it for me on Strafe.

ccMSC, 2015-10-14:
I'm guessing this is the same issue as #122. Could you follow the instructions I linked there and post the results?

rbf192, 2015-10-15:
ckb.err.log
[E] iterate_devices (usb_mac.c:362): Failed to create device plugin: e00002c7

I had nothing on ckb.log until I stopped the daemon 
ckb.log had:
    ckb: Corsair RGB driver beta-v0.2.1
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K70 RGB Gaming Keyboard  (S/N: 1002102CAE3D8C875391FD58F5001944)

[I] Caught signal 15
[I] Removed device path /var/run/ckb0

ccMSC, 2015-10-15:
Hm. This has been reported before, looks like it's the same error. Unfortunately I've never been able to reproduce it on my own computer.

Since it's only happening at startup I'd guess that it's just connecting the device too early and I need to implement a timeout/retry for it. I'll have a testing version ready tomorrow.

Could you try deleting ckb.app temporarily to confirm that the issue only exists when the driver is present? I don't *think* that the base OSX driver has startup problems the way that Linux does, but I'd like to be sure.

rbf192, 2015-10-16:
it happened even without ckb installed :(

ccMSC, 2015-10-19:
Can anyone else with the issue confirm this? I've looked into it but unfortunately, if this is the case, there's nothing I can do about it. It's possible you might be able to fix it by using different ports or a USB hub, but results with those will vary.

Hant-Teur, 2015-11-10:
yup i've got the same problem

shawjak3, 2015-11-16:
I just posted an issue numbered 160. I uninstalled the ckb.app and unplugged my keyboard, also a k65 rgb, and plugged it back in. The lights are all red but the keyboard works, the only way i get the binding and lighting affects from ckb.app is if I totally reinstall ckb every time i disconnect my keyboard and connect it again. Reboots sometimes fix it but its usually 2-3 reboots that does it.

theboasnake, 2015-12-16:
I had the same problem with my K95 RGB, this solved it:
-Don't set ckb as a startup service during installation (remove it if you already set it)
-Start Mac with keyboard in BIOS mode (This may be an extra step, made hackintosh boot way faster)
-Put keyboard back in normal mode and log in
-Start daemon manually

closed 2016-01-15 by ccMSC