ccMSC/ckb #140: Install issue.

Thunder-Child, 2015-10-13:
This is now the 3rd time I am installing this software (for various reasons) and unfortunately I can't seem to figure out the current issue.  I am running CentOS 6.4 2.6.32-358.11.1.el6.x86_64, Qt5.1.1 (also tried with 5.5), libudev-147-2, zlib-1.2.3-29, gcc-4.4.7-3, gcc-c++-4.4.7-3 in lieu of g++, and glibc-2.12-1.

The error I am getting starts off:

`make[2]: Leaving directory `/usr/local/ckb/ckb   /src/ckb'                                                                                                                                           
make[1]: Leaving directory `/usr/local/ckb/ckb/src/ckb'                                                                                                                                           
kbbindwidget.ui: Warning: Z-order assignment: 'verticalSpacer' is not a valid widget.                                                                                                             
device_vtable.c:29: error: unknown field ‘hwload’ specified in initializer                                                                                                                        
device_vtable.c:29: warning: initialization from incompatible pointer type                                                                                                                        
device_vtable.c:30: error: unknown field ‘hwsave’ specified in initializer                                                                                                                        
device_vtable.c:30: warning: excess elements in union initializer                                                                                                                                 
device_vtable.c:30: warning: (near initialization for ‘vtable_keyboard’)`

and goes for quite some time.  This might have something to do with libudev since I forgot to install it and was getting:

`make[2]: Leaving directory `/usr/local/ckb/ckb/src/ckb'                                                                                                                                           
make[1]: Leaving directory `/usr/local/ckb/ckb/src/ckb'                                                                                                                                           
In file included from includes.h:4,                                                                                                                                                               
                 from device.h:4,                                                                                                                                                                 
                 from devnode.c:1:                                                                                                                                                                
os.h:30:21: error: libudev.h: No such file or directory                                                                                                                                           
In file included from includes.h:4,                                                                                                                                                               
                 from device.h:4,                                                                                                                                                                 
                 from input_mac.c:1:                                                                                                                                                              
os.h:30:21: error: libudev.h: No such file or directory`

instead.  There are things that get thrown after that, but they all appear to be warnings and not errors.  The errors terminate at:

`make[2]: *** [release/device_vtable.o] Error 1`

The script as a whole terminates at `make: *** [sub-src-ckb-make_first] Error 2`

Let me know if you have any questions.

Thank you.

skandalfo, 2015-10-13:
The error mentions input_mac.c, but you mention you're on centos. It looks like somehow the make system thinks you're on MacOS X?

Thunder-Child, 2015-10-13:
Well, I am not getting that error any more after installing lubudev, and it might have been part of the search for said lib.  I was going to attach my Makefile and make output, but github seems to hate my attachments so I will paste them inline.

`#############################################################################
# Makefile for building: ckb
# Generated by qmake (3.0) (Qt 5.1.1) on: Tue Oct 13 10:00:47 2015
# Project:  ckb.pro
# Template: subdirs
# Command: /opt/Qt5.1.1/5.1.1/gcc_64/bin/qmake -o Makefile ckb.pro
#############################################################################

MAKEFILE      = Makefile

first: make_first
QMAKE         = /opt/Qt5.1.1/5.1.1/gcc_64/bin/qmake
DEL_FILE      = rm -f
CHK_DIR_EXISTS= test -d
MKDIR         = mkdir -p
COPY          = cp -f
COPY_FILE     = cp -f
COPY_DIR      = cp -f -R
INSTALL_FILE  = install -m 644 -p
INSTALL_PROGRAM = install -m 755 -p
INSTALL_DIR   = $(COPY_DIR)
DEL_FILE      = rm -f
SYMLINK       = ln -f -s
DEL_DIR       = rmdir
MOVE          = mv -f
SUBTARGETS    =  \
		sub-src-ckb-daemon \
		sub-src-ckb \
		sub-src-ckb-ripple \
		sub-src-ckb-wave \
		sub-src-ckb-gradient \
		sub-src-ckb-pinwheel \
		sub-src-ckb-random \
		sub-src-ckb-rain

sub-src-ckb-daemon-qmake_all:  FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile
	cd src/ckb-daemon/ && $(MAKE) -f Makefile qmake_all
sub-src-ckb-daemon: FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && $(MAKE) -f Makefile
sub-src-ckb-daemon-make_first: FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && $(MAKE) -f Makefile 
sub-src-ckb-daemon-all: FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && $(MAKE) -f Makefile all
sub-src-ckb-daemon-clean: FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && $(MAKE) -f Makefile clean
sub-src-ckb-daemon-distclean: FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && $(MAKE) -f Makefile distclean
sub-src-ckb-daemon-install_subtargets: FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && $(MAKE) -f Makefile install
sub-src-ckb-daemon-uninstall_subtargets: FORCE
	@test -d src/ckb-daemon/ || mkdir -p src/ckb-daemon/
	cd src/ckb-daemon/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb-daemon/ckb-daemon.pro -o Makefile ) && $(MAKE) -f Makefile uninstall
sub-src-ckb-qmake_all:  FORCE
	@test -d src/ckb/ || mkdir -p src/ckb/
	cd src/ckb/ && $(QMAKE) /usr/local/ckb/ckb/src/ckb/ckb.pro -o Makefile
	cd src/ckb/ && $(MAKE) -f Makefile qmake_all
sub-src-ckb: FORCE
	@test -d src/ckb/ || mkdir -p src/ckb/
	cd src/ckb/ && ( test -e Makefile || $(QMAKE) /usr/local/ckb/ckb/src/ckb/ckb.pro -o Makefile ) && $(MAKE) -f Makefile
sub-src-ckb-make_first: FORCE
[... 679 lines cut ...]
device_vtable.c:153: warning: excess elements in union initializer
device_vtable.c:153: warning: (near initialization for â€˜vtable_mouseâ€™)
device_vtable.c:154: error: unknown field â€˜allocprofileâ€™ specified in initializer
device_vtable.c:154: warning: excess elements in union initializer
device_vtable.c:154: warning: (near initialization for â€˜vtable_mouseâ€™)
device_vtable.c:155: error: unknown field â€˜loadprofileâ€™ specified in initializer
device_vtable.c:155: warning: excess elements in union initializer
device_vtable.c:155: warning: (near initialization for â€˜vtable_mouseâ€™)
device_vtable.c:156: error: unknown field â€˜freeprofileâ€™ specified in initializer
device_vtable.c:156: warning: excess elements in union initializer
device_vtable.c:156: warning: (near initialization for â€˜vtable_mouseâ€™)
device_vtable.c:157: error: unknown field â€˜updatergbâ€™ specified in initializer
device_vtable.c:157: warning: excess elements in union initializer
device_vtable.c:157: warning: (near initialization for â€˜vtable_mouseâ€™)
device_vtable.c:158: error: unknown field â€˜updateindicatorsâ€™ specified in initializer
device_vtable.c:158: warning: excess elements in union initializer
device_vtable.c:158: warning: (near initialization for â€˜vtable_mouseâ€™)
device_vtable.c:159: error: unknown field â€˜updatedpiâ€™ specified in initializer
device_vtable.c:160: warning: excess elements in union initializer
device_vtable.c:160: warning: (near initialization for â€˜vtable_mouseâ€™)
make[2]: *** [release/device_vtable.o] Error 1
make[2]: Leaving directory `/usr/local/ckb/ckb/src/ckb-daemon'
make[1]: *** [release] Error 2
make[1]: Leaving directory `/usr/local/ckb/ckb/src/ckb-daemon'
make: *** [sub-src-ckb-daemon-make_first] Error 2`

ccMSC, 2015-10-14:
Someone brought up this problem before and mentioned that using a newer version of GCC fixed it. I don't know if there's any other workaround for it.

Thunder-Child, 2015-10-15:
Yea, that did the trick, although I shudder to think of how many other things I broke.  Thanks.

closed 2015-10-15 by Thunder-Child