ccMSC/ckb #141: Mac El Capitan Build w/ Strafe RGB Build Help

rinkelj, 2015-10-13:
I had trouble getting the Mac OS version to work with El Capitan and I thought I would write some instructions for those that may be trying to test the same build on a Mac.
Pre requisit follow all build instructions here for Mac.
-After downloading the package from Test, open the ckb.pro file in QT creator
- Do a search and replace for macosx10.10 and replace with macosx10.11
  (Mine had 9 replacements made)
- under ckb-daemon open the source file usb_mac.c
- at line 399 change from:
  int products[] = { P_K65, P_K70, P_K70_NRGB, P_K95, P_K95_NRGB, P_M65, P_STRAFE_RGB };

    to: 

  int products[] = { P_K65, P_K70, P_K70_NRGB, P_K95, P_K95_NRGB, P_M65, P_STRAFE_NRGB,P_STRAFE };

- Close the project from QT creator and Save all files
- Launch quickinstall normally and continue per the instructions provided

I was able to successfully build and launch and now I am testing the solution.

Thank you for the great work.

closed 2015-10-14 by rinkelj

ccMSC, 2015-10-14:
Should be fixed proper as of f24b7d84830ce0e8f3bc308472bab17dd72e4e3d. I removed the references to `QMAKE_MAC_SDK` altogether since they were a workaround for a problem that doesn't seem to exist anymore. Let me know whether the current version builds for you.

rinkelj, 2015-10-14:
Verified