ccMSC/ckb #143: Keyboard layout lost problem

jkaart, 2015-10-19:
It not remember my layout what I have chose after close settings app.
Are this known bug or not?

ccMSC, 2015-10-21:
Are the other settings (lighting, binding, etc) remembered correctly?

jkaart, 2015-10-21:
No, other settings also resetting!

ccMSC, 2015-10-24:
Looked into it, seems the settings weren't being automatically synced like they were supposed to. See if the latest testing version fixes the problem.

jkaart, 2015-10-26:
No, it not fix this problem.
I research this my problem and it seems happen when I reboot my computer.

Btw my keyboard is K70 RGB and OS is 64 bit Ubuntu 14.04 Mate Edition if it help?!

ccMSC, 2015-10-28:
Hmm. Someone reported a similar problem in the past, but it was never resolved: #71

I've updated [the settingtest branch](https://github.com/ccMSC/ckb/tree/settingtest) so it's in sync with the latest testing version. Please install it using the `quickinstall` script, then run in a Terminal: `/usr/bin/ckb >~/ckb.log`

This should produce a file called `ckb.log` in your home directory. Upload it here and I'll see if I can find out what's going on.

meduzapat, 2015-10-28:
I have the same keyboard and similar distro, and Is working fine for me. are you sure you are pressing send to hardware button?

jkaart, 2015-10-28:
I put ckb.log to pastebin because I don't know how I post it there
http://pastebin.com/nAav1n5w

ccMSC, 2015-11-07:
Ok, it seems it did in fact save something at one point, but is failing to save anything after that. Maybe try deleting the `ckb` folder in your `~/.config` directory?

I've done some googling to see if this is a known Qt issue, but that doesn't seem to be the case, so I'm not really sure where to go from here.

jkaart, 2015-11-08:
At this ckb folder are root user rights when I delete this folder and now ckb generate folder with normal user rights.
Now my problem looks like solved

Thanks!

closed 2015-11-08 by jkaart