ccMSC/ckb #155: SCIMITAR RGB Scrolling issue

ctgswine, 2015-11-12:
SCIMITAR RGB wheel up and wheel down are bound correctly but didn't work.it look like same issue as #144 (it works regularly in windows).Also,special button is only 4-8,that's not enough for this mice.Thanks for you great work on this app!

ccMSC, 2015-11-13:
Which OS? Does the wheel work sometimes but not all the time, as described in the post you referenced?

Special buttons stop at 8 because the OS simply does not recognize more mouse buttons than that. You're free to bind the remaining buttons to keyboard keys or other functions.

ctgswine, 2015-11-14:
OSX10.11,El Capitan.Well,it actually works,but rolling 20-30times like 1 roll,so you have to roll it very hard.And i uninstalling this drive it can work functionally.Maybe is system compatibility issues?

ccMSC, 2015-11-16:
Possibly a problem with scroll acceleration? Try going to your mouse settings in system preferences and adjusting the scrolling speed, see if that solves it.

samcat116, 2015-11-16:
The M65 RGB acts in the same exact manner that he described. I tried playing with all the mouse settings and nothing seems to have made a difference.

ccMSC, 2015-11-18:
Hmm. It sounds like it's a problem with the scroll calculations so I'll try adding a workaround to turn it off. I'll let you know when it's ready.

ccMSC, 2015-12-07:
As of c3474d2ec359b429b402d79a58b74fba9938cfc2 it's now possible to disable scroll acceleration from the GUI. You can access it under "OSX tweaks" in the "More settings" screen. Once enabled, the scroll wheel should behave consistently.

Let me know if you need a pre-compiled version to test with.

closed 2016-01-15 by ccMSC

moroten1337, 2016-08-28:
thank you SO much for fixing this <3