ccMSC/ckb #156: Missing keycodes to be able to map

OvermindDL1, 2015-11-14:
Linux systems support a vastly larger amount of key bindings than what is currently supported by the application.  Would it be easily possible to be able to supply our own scancodes?  Specifically I am wanting to map the [XF86](http://www.xfree86.org/4.0/xmodmap.1.html) keycodes as I used them on my old keyboard to the G1-G18 keys on my K95 replacement keyboard, but the application does not seem to allow defining your own scancodes (of which the XF86 extended keys are the most common set).  I am not looking for the XF86 keys specifically but rather just anything unique and unused, thus even the native F13+ and others would work for me quite well.  Having full support for the entire scancode range supported by xevent would of course be perfect.  At the moment I am using xdotool to send the wanted key when pressing the physical key, however that is adding a noticable latency that would not exist if the program were able to send the full key code range supported by xevent instead.  If it would not be too involved to do I would send such a patch if you can point me to the basics of the code structures, documentation, or anything of the sort?

ccMSC, 2015-11-16:
Not currently planned since most of them would have no OSX equivalent and I try not to implement features that behave inconsistently across operating systems. 

If you want to have a go at it yourself, most of the relevant code would be in `src/ckb-daemon/input.c`

closed 2015-11-23 by ccMSC

t0mm1, 2015-11-27:
What about F16-F19 which are included on the Apple Keyboard with Numeric Keypad? Would be nice to include them. IIRC the windows driver lets you choose from a range of F1-F25.

I don't feel confident enough in coding, so I can't fork and send a pull request, so I just wanted to leave my voice here.

ccMSC, 2015-11-29:
@t0mm1 The driver actually does have support for some extra Fn keys, but apparently I forgot to add them to the GUI. I'll address that in the next few days.

bduong, 2015-12-02:
I noticed that some of the other Apple Keys are included in the keymap but aren't mappable in the UI (specifically eject and power). Are those also supported in the driver?

Would be nice to hit the sleep shortcut (Ctrl + Shift + Eject/Power)