ccMSC/ckb #158: Keymap issue

OvermindDL1, 2015-11-14:
I have a keyboard-like device, Naga mouse, that I set a custom keymap on via xkbcomp, been working for me for years.

However now that I have hooked up the K95 with ckb randomly within an hour after bootup then its keys break.  Specifically I change the Naga's 1,2,3,4,5,6,7,8,9,10,11,12 keys to be F13 through F24, and this keymap is applied at bootup and it and the K95 keyboard work as expected, however about 5-10 minutes after bootup the K95 keyboard seems to get its keys at the top of the keyboard, not the keypad as the Naga maps to, get remaped to F13 to F24 as well, yet the numpad continues to work fine, so that is a disconnect, and it is especially odd as it happen a while after xkbcomp is ever run.

So I started looking, the first thing I notice is that the K95 is not appearing via the xinput list:
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Razer Razer Nostromo                      id=9    [slave  pointer  (2)]
⎜   ↳ Razer Razer Naga 2014                     id=11   [slave  pointer  (2)]
⎜   ↳ ckb1: Corsair K95 RGB Gaming Keyboard     id=13   [slave  pointer  (2)]
⎜   ↳ Razer Razer Naga 2014                     id=10   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Razer Razer Nostromo                      id=8    [slave  keyboard (3)]
    ↳ Razer Razer Naga 2014                     id=12   [slave  keyboard (3)]
```
It is not listed in the Virtual core keyboard whatsoever, only under the Virtual core pointer for the mouse emulation I am guessing, so I cannot even apply an xkbcomp keymap at all to fix it, I have to reboot to fix it, for only a few minutes at that.

If I do not have xkbcomp set a keymap on the naga at bootup then the keyboard's mapping never changes.

Thus, how is the mapping ever applied to the K95 whatsoever when it is not appearing in `xinput --list` at all?

Second, how is the K95 not appearing in `xinput --list`?

My initial hypothesis based on how it is working is that it is using the last-most listed device in Virtual core keyboard as its mapping and sending the key inputs as if it were from that device instead of its own device.  This is further supported by I just tested by forcing a reshuffle of the ID's of the Naga so there is a keyboard higher than it then the keyboard works (I did this by plugging in my old keyboard, which caused it to appear with id=14, and as evidenced by my parenthesis working again).

It seems everything comes down to the fact that there is no keyboard device setup for the K95 in the xinput system, so how is it sending events unless via another existing device by mocking it?

OvermindDL1, 2015-11-14:
And after unplugging the keyboard I plugged in, less than an hour later the K95 reverted back to the broken keymap, still not showing in `xinput --list`.

ccMSC, 2015-11-16:
Take a look at this post: http://forum.corsair.com/v3/showpost.php?p=817243&postcount=543
Not sure it's exactly the same issue but it would seem to be related.

Given how buggy Xorg seems to be in handling combined keyboard+mouse devices, I'll probably just update the driver to create two separate devices for it...

OvermindDL1, 2015-11-20:
That is what was done for the G15 I used to use and it worked well.

Cannot wait for Wayland...

ccMSC, 2015-11-23:
As of 258988b2b9303815929cec02c7e0957a3e13020d the inputs are now split into two separate devices. When I use `xinput list` I see one under keyboard and one under mouse. Hopefully this fixes your issue.

OvermindDL1, 2015-12-30:
Testing the 'testing' branch, but still only getting:
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ckb1: Corsair K95 RGB Gaming Keyboard     id=8    [slave  pointer  (2)]
⎜   ↳ CHICONY USB Keyboard                      id=9    [slave  pointer  (2)]
⎜   ↳ Razer Razer Naga 2014                     id=11   [slave  pointer  (2)]
⎜   ↳ Razer Razer Naga 2014                     id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Razer Razer Naga 2014                     id=10   [slave  keyboard (3)]
    ↳ CHICONY USB Keyboard                      id=12   [slave  keyboard (3)]
```
And after a few keystrokes the Corsair keys get remapped to the same as the Naga keys, yet the old keyboard (the Chicony thing) is not being remapped, only the Naga is being remapped via xkb, and the Corsair is not appearing in the core keyboard section.

OvermindDL1, 2015-12-30:
The especially odd thing, when xkbcomp'ing a new mapping onto the naga it gets the change immediately, but the Corsair keyboard retains its original mapping for about one minute, then it gets changed to the same mapping as the Naga (and ditto on the minute when reverting the mapping back to the original).  The old Chicony keyboard remains on its own mapping set with the keys working properly the entire time.

OvermindDL1, 2015-12-30:
I wiped out the existing files and rebuilt and it seems to be working better, two devices now, testing to make sure no mapping 'leaking'...

OvermindDL1, 2015-12-30:
Testing for about 10 minutes in a variety of contexts shows that it works and the mappings now stick.  You are amazing, thanks for the work!  I'd say this is up for closing.  ^.^

closed 2016-01-15 by ccMSC