ccMSC/ckb #160: Corsair K65 RGB not staying connected

shawjak3, 2015-11-16:
I use this awesome program on my macbook when playing games or coding and its hooked up to another monitor. I also use it at work because its a dream to code on, however when disconnecting the keyboard to play a game on my macbook and then connecting it back to my iMac at work it doesn't ever recognize it. Sometimes a reboot fixes it but I'm wondering if there is something I need to do for constant connectivity. I am on version 2.2.

ccMSC, 2015-11-18:
Assume you're running the latest driver on both? Maybe try using a different port on the iMac or flipping the poll rate switch near the back?

shawjak3, 2015-11-18:
That seems to do it, if I move that poll switch around and then back it connects to it again. Thanks for the help!

closed 2015-11-23 by ccMSC