ccMSC/ckb #162: ~` key types like a ±§ key on OS X

BenLeggieroNCR, 2015-11-17:
Before I installed _ckb_, every key typed just fine. Now, every key types just fine except the one to the left of the <kbd>!<br/>1</kbd> key. Now, that key is acting as if it is a <kbd>±<br/>§</kbd> key (typing it results in `§`, and holding shift and typing it results in `±`). Quitting _ckb_ does not fix this, nor does restarting it.

BenLeggieroNCR, 2015-11-17:
Turns out it defaults to European keyboards. Strange, since I told OS X it was American layout before installing ckb. Consider this issue that **ckb ignores keyboard locale on install.**

ccMSC, 2015-11-18:
Not sure it's possible to fix that, unfortunately. It should ask you to specify your keyboard layout the first time you run it (as of v0.2.2 - if you installed an earlier beta, it wouldn't).

BenLeggieroNCR, 2015-11-18:
Awww, sad. You can't ask OS X for its current locale? :/

ccMSC, 2015-11-21:
There is a locale detection which works on Linux and it used to work on OSX, but I think it stopped working with Yosemite. I'll revisit it in the future, but for now it's not a high enough priority.

soybeanyoung, 2015-11-22:
I suffered from the same issue, but I found we could change the layout in the settings of ckb.

[image]

The default one is European, so that is why we have such problems.

darkcrux, 2015-11-25:
Oh that worked! was pulling my hair looking for a fix! Thanks!

closed 2015-11-25 by BenLeggieroNCR