ccMSC/ckb #163: High cpu load

soybeanyoung, 2015-11-18:
If 'show animated' is checked, the driver will consume up to 70% cpu. The way to solve it is to uncheck this box and extend animation duration. However, in this case, it will still occupy up to 10% cpu sometimes.

If I uninstall the driver, a lot of keys including space and backspace do not work. Even though I deleted /Library/LaunchDaemons/com.ckb.daemon.plist manually, this would happen, too. Is there any other configuration files? When I reinstalled the driver, I found my profile was loaded automatically.

ccMSC, 2015-11-18:
CPU usage: see #67 

There are some configuration files in `~/Library/Application Support` but they only serve the GUI application. Without the driver installed they won't do anything.

soybeanyoung, 2015-11-18:
Thanks. I will check the issue regarding cpu usage.

In terms of the other issue, that was my own fault, cause I enabled a configuration file in usb overdrive( I need that software to remap the other keyboard). Sorry for the disturbance.

ccMSC, 2015-11-23:
Ok. Closing this issue for now - keep an eye on the previously-mentioned issue for updates about CPU usage.

closed 2015-11-23 by ccMSC