ccMSC/ckb #164: saving animations

ray-420, 2015-11-18:
I'm running ubuntu 15.10 with the Corsair k70 rgb
ckb wont save  animations and i'm having problems when my computer goes into suspend mode.
while in suspend the leds stay illuminated and key presses wont bring computer out of suspend. 
If i run ckb as root it allows me to save animations to keyboard and the leds go ouit during suspend, but the app indicator doesn't show when ckb is run with sudo. Not sure if it would help the suspend but if you maybe did a password prompt while saving to hardware. Im not a programmer so i could be way off but i have it starting up with sudo at boot and its working great except my app indicator isnt working.

ccMSC, 2015-11-18:
The GUI shouldn't be run as root. You might need to check the permissions of the `.config` folder in your home directory, and/or try removing `~/.config/ckb` and then re-running the application.

ray-420, 2015-11-18:
what should they be set at

ray-420, 2015-11-18:
cool thanks that fixed my problems didn't think of that. I thought maybe it was saving something in system to change that and is why it worked with sudo but apparently it gave ther config file to root when running the install file.

ray-420, 2015-11-18:
I ran the install script  as root cause it wouldn't run if not so not sure if that's why or not and why it made me use root but everything is working awesome now. and thanks for supporting this keyboard. I love it

ccMSC, 2015-11-21:
Ok, sounds like that was the problem. I'll try to add some protection for that in a future version.

closed 2015-11-21 by ccMSC