ccMSC/ckb #165: Latin American Layout

carlosmprzmx, 2015-11-18:
Do you plan give support for the Latin American layout?

Actually i'm using you software in a iMac OS X 10.11.1 with a Corsair K70 RGB and run amazing!

Thank you guys!

ccMSC, 2015-11-21:
I usually add them on a per-request basis. Which layout specifically do you want?

carlosmprzmx, 2015-11-21:
Thanks for the answer.

I have the Corsair K70 RGB with the latin american layout, this -> 
[image]

By the way, you are doing a great job!

carlosmprzmx, 2015-11-21:
I decide take a picture if the keyboard.

[image]

i hope the photo be more useful

ccMSC, 2015-11-23:
Implemented in 810b8db89fd0750f65da9c29c244afd9329b1744. Let me know if you have any issues with it.

carlosmprzmx, 2015-11-28:
Instalado todo parece estar bien.

closed 2016-01-15 by ccMSC