ccMSC/ckb #169: Fixed login screen issue, but have to manually start daemon

shelbym5, 2015-11-23:
This is with k70 rgb on mac osx 10.11

I had an issue where I always had to reconnect my k70, or sometimes switch the bios switch, or sometimes both, to get it to work in the OSX login screen. Using the following plist I was able to fix that so I don't need to reconnect my keyboard or switch the bios switch every time.

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Label</key>
    <string>com.ckb.daemon</string>
    <key>ProgramArguments</key>
    <array>
        <string>/Applications/ckb.app/Contents/Resources/ckb-daemon</string>
    </array>
    <key>KeepAlive</key>
    <true/>
    <key>ProcessType</key>
    <string>Interactive</string>
    <key>LegacyTimers</key>
    <true/>
    <key>StandardOutPath</key>
    <string>/var/log/ckb.log</string>
    <key>StandardErrorPath</key>
    <string>/var/log/ckb.err.log</string>
</dict>
</plist>

(also, I'm not generating any logs for some unknown reason.)

Anyways, that somehow fixes my login screen problem, but ckb still doesn't recognize my keyboard unless I run the daemon manually using:

sudo /Applications/ckb.app/Contents/Resources/ckb-daemon

My question is, is there anyway to get the daemon to work without me having to run the command on terminal every time I restart my computer.

ccMSC, 2015-11-25:
I think the daemon's not being run at all? Not sure why though, the plist file looks valid to me. Try running:
```
sudo chown root:wheel /Library/LaunchDaemons/com.ckb.daemon.plist && sudo chmod 0644 /Library/LaunchDaemons/com.ckb.daemon.plist
```

shelbym5, 2015-11-25:
That didn't work and it broke working on login screen. So I try running it manually at this point and this is what I get:
```
iMac:~ smedina$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon 
Password:
    ckb: Corsair RGB driver beta-v0.2.2
[F] ckb-daemon is already running (PID 87). Try `killall ckb-daemon`.
[F] (If you're certain the process is dead, delete /var/run/ckb0/pid and try again)
iMac:~ smedina$ kill pid 87
-bash: kill: pid: arguments must be process or job IDs
-bash: kill: (87) - Operation not permitted
iMac:~ smedina$ sudo kill pid 87
kill: illegal process id: pid

ccMSC, 2015-11-26:
Try `launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist`

The log files should be there now since the daemon seems to be running.

shelbym5, 2015-11-27:
```
iMac:~ smedina$ launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
/Library/LaunchDaemons/com.ckb.daemon.plist: Could not find specified service
```

On the error log
```
[E] iterate_devices (usb_mac.c:364): Failed to create device plugin: e00002c7
```

and the  ckb.log is empty
Please note I also had to unplug and replug during the login screen

carlosmprzmx, 2015-11-28:
I have OS X 10.11.1 and K70 RGB and all running perfectly

ccMSC, 2015-11-29:
Interesting...so far everybody who's reported that error has stated that they still have the problem regardless of whether they use the driver or not. I'll see if there's anything I can do about it.

silasjmatson, 2015-11-30:
FWIW, I have a similar problem where I have to force quit the ckb-daemon on login, which causes it to be restarted and then my custom keybindings and lighting scheme work.

Before I do that, it just eats CPU and doesn't work.

K70 RGB, also on OS X (10.11)

@ccMSC Your last comment seems spot on. I always have to disconnect & reconnect the keyboard when not using ckb. Oddly enough, occasionally the driver works properly from boot and I don't have to restart ckb-daemon as I stated above -- in those cases, I never have to disconnect/reconnect the keyboard.

shelbym5, 2015-12-03:
I'm very stubborn and refused to believe that using a USB hub would fix my problem. Finding a random one lying around, I decided to give it a chance and that fixed my problem.

Case closed.

But just out of curiosity, would you happen to know why this fixed the problem? I just a curious person.

ccMSC, 2015-12-04:
As far as I can tell it's some combination of the keyboard hardware, the USB controller, and the operating system. Apparently it's a common issue on Windows as well. Corsair have attempted to fix it with firmware updates but they don't seem to have made much progress.

ccMSC, 2015-12-22:
If anybody here's still having the issue using the driver at startup, I have a possible fix in the testing repository right now. It will try to wait around for a few seconds if the setup fails the first time. Not sure if the fix will solve it or not, but it's the only idea I have at this point.

rykugur, 2016-01-01:
For what it's worth, I'm having the same/a similar issue on arch linux. I have the service set to auto-start, and use slim for my login manager (which is also auto-started). For now, I just have the auto-start disabled and start it manually after login (alternatively, ssh into this box from my laptop and kill the service, login, restart service).

I'll keep an eye out for ccMSC's upcoming fix to see if that takes care of it.

ccMSC, 2016-01-04:
The fix is OSX-only unfortunately. I don't think it's possible to solve the boot up issue on Linux aside from the steps mentioned in the Troubleshooting section.

closed 2016-01-15 by ccMSC