ccMSC/ckb #171: Mouse support

Astaoth, 2015-11-28:
Hi,

Do you plan to support also gaming mouses ? Like the Corsair M90 ?

Best regards

carlosmprzmx, 2015-11-28:
@Astaoth Corsair M90 don't have RGB lights, this software is for control RGB lights :confused:

Astaoth, 2015-11-28:
In the README, in the "Usage" section, there is written : 
"- Customizable key bindings
- Per-key lighting and animation
- Reactive lighting
- Multiple profiles/modes with hardware save function
- Adjustable mouse DPI with ability to change DPI on button press"

It seems a little more than only RGB light control.

On the Corsair M90, there are lights, internal memory, extra keys and adjustable DPI. Maybe it is possible to control this feature from this software ?

ccMSC, 2015-11-29:
Non-RGB support hasn't really been possible since they use a different USB protocol (only exception being the STRAFE) and in some cases they don't work the same way that the RGB versions do. Unless you or somebody else has the ability to add code for it, it's not likely that the M90 will get supported.

Astaoth, 2015-11-29:
Oh, I see :(. I'm not sure I have the skills to decode the USB protocol, maybe can you give me some advice ?

ccMSC, 2015-12-03:
The way I did it was to run Windows in a virtual machine, connect the keyboard/mouse to it, then capture the USB packets with Wireshark. After that it's just a matter of testing the inputs and outputs until you figure out which packet does what.

Astaoth, 2015-12-03:
Thank you, I will try to figure out the protocol.

closed 2016-01-15 by ccMSC