ccMSC/ckb #172: ckb 0.22 disconnect k95 RGB

Keridos, 2015-12-01:
On one of my PC I have a weird issue. I have my K95 RGB connected via a USB hub and a KVM switch to my PC's. One one of them I can use the K95 fine, but the other shows a log like this when trying to type:
```
    ckb: Corsair RGB driver beta-v0.2.2
[I] Root controller ready at /dev/input/ckb0
modprobe: FATAL: Module uinput not found.
[W] usbmain (usb_linux.c:403): Failed to load uinput module
[I] 8 Corsair K95 RGB Gaming Keyboard 
[I] Connecting Corsair K95 RGB Gaming Keyboard (S/N: 09005021AE3E18065487E4A2F5001944)
[I] Starting input thread for /dev/input/ckb1
[E] os_usbrecv (via led_keyboard.c:207): Connection timed out
[W] _start_dev (device.c:45): Unable to load hardware profile
[I] Corsair K95 RGB Gaming Keyboard ready at /d
```

Both have the uinput flag in kernel set and the USB HID quirks. When typing, it accepts one keypress, and then spams the keypress for a few seconds.

ccMSC, 2015-12-03:
Any idea why it's saying `Module uinput not found`? Do you get the same result if you run `sudo modprobe uinput` manually?

Keridos, 2015-12-03:
The module is built into my kernel, not a module.

ccMSC, 2015-12-04:
Does it work correctly if you put the keyboard in BIOS mode?

You might try searching the Corsair forums for "KVM" as I've seen a few Windows users struggle with this as well. I think a few people managed to get it working but it seems like the keyboard doesn't work very well with them.

Other question: Do both computers have the GUI running? The keyboard behaves slightly differently when it's under software control. If the GUI is running on one computer it probably needs to be running on both of them.

Keridos, 2016-01-01:
Tried it on BIOS mode as well, happens then, too. Also running it at both computers did not really help either.

ccMSC, 2016-01-04:
Not sure there's a solution for this, honestly. Corsair haven't made any effort to support KVM configurations and I don't have one to test with myself, so I don't have any way of debugging the problem.

Keridos, 2016-01-05:
Does Corsair offer you any support at all? Sad that this is this way.

closed 2016-01-15 by ccMSC

Keridos, 2016-05-01:
Just to let you know; I derped in the kernel config, USB devices had issues in general. Sorry for annoying you with this :-/