ccMSC/ckb #173: CKB wont open and menu bar is gone

Mikeeeey, 2015-12-04:
- Using a K65 
- Running El Capitan 10.11.1

Downloaded CKB, changed the lights on my K65 and worked fine, went to settings and said to hide the menu bar icon and now I can never open it back up, I go to Applications -> CKB and it still won't open, I've tried terminate the process and re-open but still nothing, tried reboots but nope.

Lastly, uninstalled it completely and reinstalled it, same thing it remembered my profiles though and I can't get the tray bar icon back up or to open CKB at all.

Any suggestions? 

PS - I know you've heard this a lot, but again THANKS for developing this, Corsair needs to pick up their act.

ccMSC, 2015-12-04:
You may need to delete the settings entirely. They can be found in `~/Library/Application Support/com.ckb.ckb.plist`. Make sure you've closed ckb via System Monitor first.

Mikeeeey, 2015-12-04:
There is no com.ckb.ckb.plist in my ApplicationSupport folder? 

When I terminate it from Activity monitor it just pops right back up.

ccMSC, 2015-12-05:
Sorry, it's actually in `~/Library/Preferences`, not Application Support.

helloworld82, 2016-01-06:
I'm having the same issue, and the com.ckb.ckb.plist file isn't in the Application Support folder.

closed 2016-04-08 by ccMSC