ccMSC/ckb #175: Scrolling

prototoon, 2015-12-10:
I have the same scrolling issues with my m65, I've seen two other threads that have the same problem but when i read them i totally have no idea how to fix it? Could i get some help? thanks

ccMSC, 2015-12-11:
There's a fix for it currently in the testing repository. I can provide a binary if you don't want to compile it from source.

prototoon, 2015-12-13:
that would be great :) please do, thanks

ccMSC, 2015-12-20:
Hi, sorry for the delay, I have the binary version ready: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20beta%20v0.2.2%2Bt04.zip

There's a new "More Settings" screen which has an option to disable scroll wheel acceleration. Try checking that option and see if the problem is solved.

prototoon, 2015-12-28:
thanks so much :)

prototoon, 2015-12-28:
its working perfectly now

closed 2016-01-15 by ccMSC