ccMSC/ckb #182: CKB daemon icon is on top of the 'x' used to close windows

chrisgogreen, 2015-12-20:
After installing on Ubuntu 14.04 I noticed that the CKB daemon icon is always on top of the close window x, I have to move the window to close them.

tatokis, 2015-12-20:
If I understood this correctly, do you have the libappindicator-dev package installed?
If not install it and recompile ckb.

chrisgogreen, 2015-12-20:
First I restarted and the ckb tray icon disappeared completely. Then I installed libappindicator-dev and recompiled, now the icon is in the correct location in the tray. Thanks!

closed 2015-12-20 by chrisgogreen

bigBearSingh, 2016-01-09:
Note that after installing libappindicator-dev and recompiled, I had to restart the desktop for tray icon to appear correctly, or maybe this step is obvious. Thanks all.