ccMSC/ckb #183: Bind G keys to insert custom text

chrisgogreen, 2015-12-20:
How can I bind the G keys on my K95 to insert snippets of text? It looks like I could use the Launch program on keypress feature, is that the best way?

tatokis, 2015-12-20:
Only way I can think of right now is to install the xautomation package and then use the "Launch program on keypress" with xte.
Example: 
```
xte 'str Hello, World!'
```
Maybe this could be a built in feature of ckb in the future.

chrisgogreen, 2015-12-20:
I will try this and report back.

chrisgogreen, 2015-12-20:
I ended up making a bash script with:

```
#!/bin/bash

cat /home/username/code/snippets/bare_method.txt | xclip -i -selection clipboard
xte 'keydown Control_L' 'key v' 'keyup Control_L'

```

then binding the G key to Launch program on key press with:

```
/home/username/code/snippets/bare_method
```

ccMSC, 2015-12-22:
Yeah, you can also use `xdotool type 'your text here'`.

Text macros are a planned feature, but they've been bumped down quite a lot due to lack of demand. They'll be available eventually.

chrisgogreen, 2015-12-22:
Thank you for mentioning xdotool. I'll close this issue.

closed 2015-12-22 by chrisgogreen