ccMSC/ckb #185: Indicators don't emit light

kimamb, 2015-12-24:
[STRAFE RGB SCANDINAVIAN]
All of the top row buttons don't emit light.

I've tried changing the settings under top row; still no luck.
Even the corsair logo is gone!
I have problems finding the brightness key because of my dark room.

What could be the cause of this?

SicVolo, 2015-12-24:
are you talking about the brightness and the windows lock buttons, or the esc through f12 row?
Funny thing, I did not realize corsair logo was supposed to be lit. I guess we got to fix this.

kimamb, 2015-12-24:
I didn't thought that either, before I tried playing witcher on my windows partition.

ccMSC, 2015-12-27:
Check your Performance tab in the keyboard settings - it contains colors for those lights. Make sure the indicators are enabled and are set to the colors you're expecting.

SicVolo, 2015-12-27:
These don't work on my system either. Apparently the brightness button, the windows lock button and the corsair logo use USB codes unique to Strafe RGB. I just forgot about them when I wrote the code. I'll get it fixed.

zhouti1993, 2015-12-29:
same problem for my strafe RGB.looking forward to new update.Thanks al lot!!

SicVolo, 2015-12-30:
Fixed. Just sent a pull request to the testing branch of this repo from my fork. @ccMSC keep us updated, pls.

zhouti1993, 2015-12-30:
i don't know how to do i.Can u teach me to operate more specifically? Thank u very much!

zhouti1993, 2015-12-30:
In order to install the new program,Do i  have to install Qt5?

zhouti1993, 2015-12-30:
Finally I solved out˜~  Wow amazing
  Guines@!! Engineering

zhouti1993, 2015-12-30:
HAHA  works perfect!!!

kimamb, 2015-12-30:
@SicVolo Thank you!

ccMSC, 2015-12-31:
Merged changes into testing, thanks!

kimamb, 2015-12-31:
The corsair logo somehow became white after rebooting, in all of the profiles I changed its colour in.

SicVolo, 2015-12-31:
That's probably because it had not had a color 'saved' in the past. Try setting it to a color in CKB and use "save to hardware" button to have it default to that color when booting, or when not controlled by CKB. Mine seems to work fine (boots red). The indicator color can be controlled via the performance tab.

kimamb, 2015-12-31:
Ah

shazron, 2016-01-09:
Works great for me from the testing branch, thanks @SicVolo [Strafe RGB US]

closed 2016-01-15 by ccMSC