ccMSC/ckb #187: M65 RGB mouse not being detected

Benworkinger, 2015-12-25:
I just got the M65 RGB mouse for xmas. I have a Mac but no matter what, my mouse isn't recognized. What do I do?

Thanks

ccMSC, 2015-12-27:
Is the mouse picked up by the system itself (i.e. input works, LEDs light up) or is it completely non-responsive?

Sometimes it's necessary to perform a firmware update from CUE the first time you get a new device. If you have a Windows computer available, try connecting it to that first.

Benworkinger, 2015-12-27:
yes, the mouse part works on my mac. It just isn’t being recognized by your program.

ccMSC, 2016-01-04:
I've made some changes in the last few testing version which might be able to fix the problem.

Here's a binary version if you don't want to compile the source: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20beta%20v0.2.2%2Bt07.zip

Benworkinger, 2016-01-04:
Ok,I connected the mouse to a windows computer and it worked fine. I then connected the mouse back up to my mac, with the latest ckb program and the cab still says "no device connected". The mouse works as a normal mouse and the LEDs light up but I cannot change the color or rebind the keys.

ccMSC, 2016-01-05:
Could you follow these instructions to get a log from the driver: https://github.com/ccMSC/ckb/issues/106#issuecomment-131287945

The first two lines are Terminal commands to run the driver manually. If that produces the error, copy the output and paste it here. If it sets up the mouse successfully, move on to the second part.

closed 2016-01-15 by ccMSC