ccMSC/ckb #192: Hi all

Omerald, 2016-01-01:
I am owner of a Corsair rgb scimitar mouse.
It's all right but playng Diablo 3 my cursor continuously tremble,
I don't know what can i do to stop it. Thanks

a2aaron, 2016-01-02:
This probably isn't a problem with ckb. Check what DPI you've set the mouse to, what mouse pad you're using, and if mouse acceleration is turned on in your setting.

Omerald, 2016-01-03:
in this pic you can see dpi settings of my Corsair Gaming SCIMITAR RGB MOBA/MMO Gaming Mouse .
Trembling is displayed only  when i move the mouse. When i work in desktop  also can see pointer move to the angle on the upper left. Yes I know the issue is more of one ;((
Thank you for your replay.
[image]

Omerald, 2016-01-03:
Where can I adjust the mouse acceleration? Isystem preference or with ckb?

a2aaron, 2016-01-04:
You can enable/disable mouse acceleration in your system preferences (you may need to look online)

ccMSC, 2016-01-04:
Also check to make sure that your firmware is up to date and check to make sure the physical sensor is clean. I've heard that some of the Scimitars had issues like this, affecting Windows users as well. Supposedly one of the firmware updates fixed it.

Omerald, 2016-01-06:
I am following your suggestions but unfortunately without solving any issue. 
When I am playng Diablo 3 , moving cursor the pointer tremble, only in diablo 3 !!!
I am trying to contact Blizzard assistance , maube they can help me.
Thanks for your help, I appreciate!
If I solve it I will came back here to tell you.

closed 2016-04-08 by ccMSC