ccMSC/ckb #193: M65 Mouse Recognition Bug

a2aaron, 2016-01-01:
Quick note: I have a
- unupdated M65 mouse (never connected to a Windows),
- a iMac (Late 2012 model) with USB 3.0 hubs.
- using the ckb binaries (not building from source)

ckb doesn't seem to recognize the M65 mouse. It says "no device connected". In addition, ckb daemon seems to take up 50% of a core.

The mouse keeps colors assigned to it and default functionality still exists (sniper button, left and right clicking, scrolling, etc) but custom buttons bindings, animations, and custom DPI settings don't work without ckb working.

This is fixed/prevented in two ways: 
1. Reinstall ckb (also fixes CPU usage)
2. Keep the mouse unplugged until logged in. (doesn't seem to always work)

The computer does seem to recognize that the mouse is plugged in, but ckb doesn't.

Benworkinger, 2016-01-04:
Will connecting the mouse (and updating it) make it be recognized by ckb?

a2aaron, 2016-01-04:
No for reconnecting. I can't update the mouse because I don't have access to a windows computer to update the mouse firmware, however, I am on firmware version 1.19 for my mouse.

Benworkinger, 2016-01-04:
I had the same problem as you, if I have access to a windows computer will it fix it?

ccMSC, 2016-01-04:
Closing as a duplicate of #187. I've posted a possible solution in that thread.

closed 2016-01-04 by ccMSC