ccMSC/ckb #208: CKB wont open on OSX Mountain Lion

orishiku, 2016-01-05:
Hi, first of all: i love this project, I'm using it on linux with my scimitar rgb and run like a charm. Recently i had to use mac (crying in spanish), but after i install the binary or compile the source, the ckb.app won't open. I'm using OSX 10.8.5
thanks for all your effort :3

i tried to run it from console and i get this:

/Applications/ckb.app/Contents/MacOS/ckb
2016-01-04 23:43:26.934 ckb[14653:707] -[NSProcessInfo beginActivityWithOptions:reason:]: unrecognized selector sent to instance 0x7f859041be40
2016-01-04 23:43:26.937 ckb[14653:707] An uncaught exception was raised
2016-01-04 23:43:26.941 ckb[14653:707] -[NSProcessInfo beginActivityWithOptions:reason:]: unrecognized selector sent to instance 0x7f859041be40
2016-01-04 23:43:26.944 ckb[14653:707] (
	0   CoreFoundation                      0x00007fff8d056b06 __exceptionPreprocess + 198
	1   libobjc.A.dylib                     0x00007fff8e39b3f0 objc_exception_throw + 43
	2   CoreFoundation                      0x00007fff8d0ed40a -[NSObject(NSObject) doesNotRecognizeSelector:] + 186
	3   CoreFoundation                      0x00007fff8d04502e ___forwarding___ + 414
	4   CoreFoundation                      0x00007fff8d044e18 _CF_forwarding_prep_0 + 232
	5   ckb                                 0x0000000107b968ac main + 140
	6   libdyld.dylib                       0x00007fff8272e7e1 start + 0
)
2016-01-04 23:43:26.946 ckb[14653:707] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSProcessInfo beginActivityWithOptions:reason:]: unrecognized selector sent to instance 0x7f859041be40'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff8d056b06 __exceptionPreprocess + 198
	1   libobjc.A.dylib                     0x00007fff8e39b3f0 objc_exception_throw + 43
	2   CoreFoundation                      0x00007fff8d0ed40a -[NSObject(NSObject) doesNotRecognizeSelector:] + 186
	3   CoreFoundation                      0x00007fff8d04502e ___forwarding___ + 414
	4   CoreFoundation                      0x00007fff8d044e18 _CF_forwarding_prep_0 + 232
	5   ckb                                 0x0000000107b968ac main + 140
	6   libdyld.dylib                       0x00007fff8272e7e1 start + 0
)
libc++abi.dylib: terminate called throwing an exception
Abort trap: 6

ccMSC, 2016-01-05:
Unfortunately I'm not able to support OS X versions earlier than Mavericks.

If you're willing to debug the code and figure out which sections need to be changed in order for it to work, I can integrate the changes. I don't have the ability to test them myself.

orishiku, 2016-01-06:
thanks for the response, fortunately my mac can be upgraded exactly one more punch :V, and solves de problem :3

ccMSC, 2016-01-11:
Okay, glad you got it fixed!

closed 2016-01-11 by ccMSC