ccMSC/ckb #209: No light

LadySleepsAlot, 2016-01-05:
Hello, and thanks so much for making ckb, I love it!

I recently updated to the latest version, and one of two things happen. Either it wont find my keyboard, or it finds it but makes all the lights go out. If option two occurs and I push the light button on the keyboard, it will shut down ckb completely.

I always unplug my keyboard when rebooting, but it seems like it I plug it in after opening ckb, it will not find it, if I plug it in before, it will go completely dark.
When it finds it and makes it dark, there is no keyboard showing in the program itself where I can usually change the colors of the keys.
I'm using El Capitan 10.11.2, and updated my keyboard firmware today before installing the newest version of ckb (updated using a windows computer)

I'm not very tech savvy, and I hope I managed to explain everything well enough.

ccMSC, 2016-01-11:
Sounds like it might be a bug in the GUI. Could you try installing a newer testing version to see if it helps? Here's a download link: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20beta%20v0.2.2%2Bt07.zip

LadySleepsAlot, 2016-01-11:
That worked like a charm, thanks so much :)

closed 2016-01-15 by ccMSC