ccMSC/ckb #210: No fn(function) Key binding. Necessary for OSX.

LiranCohen, 2016-01-07:
There is no key binding for the "fn"(function) key that's frequently used on OSX.  The keyboard has an fn key and the original worked fine.

Thanks

ccMSC, 2016-01-11:
(Note: I don't actually have one of the keyboards with the Fn key, so bear with me)

Do you mean the media keys stop working when the driver is active, or something else? As far as I'm aware the Fn key doesn't do anything on its own.

shazron, 2016-01-11:
With the master code, my binding layout on my Strafe RGB shows the FN key as the "right option" key (key between Right Alt and the Menu key).

I use it as a modifier for the F5, F7-F12 keys (Media Keys).

Usually there is a checkbox in the Keyboard System Preference where if checked, you won't need to use the FN key to modify the media keys, but this is missing for this keyboard (not sure if this is an OS X thing). Not sure if its even possible to bind to Exposé or Dashboard even.

shazron, 2016-01-19:
See #223 thanks to @SicVolo

ccMSC, 2016-01-26:
Turns out I was wrong - the Fn key actually *does* do something on its own, so thanks to SicVolo's work and a few patches of my own it should be possible to use that functionality now.

Here's the latest testing binary: https://dl.dropboxusercontent.com/u/16424004/ckb/ckb%20v0.2.3%2Bt04.zip

@shazron Regarding that setting in the Keyboard preferences, it only applies to the Apple aluminum keyboards and won't work on any others. However, the shortcuts for those keys can be rebound via `System Preferences > Keyboard > Shortcuts`.

The media key bindings will remain in effect as Corsair has programmed them into the hardware and it seems that they can't be overwritten.

closed 2016-04-08 by ccMSC