ccMSC/ckb #214: OSX 10.11.2 Strafe RGB doesn't work on boot.

chris37879, 2016-01-09:
After a reboot my Strafe RGB doesn't work. CKB says no device detected, however if I rerun the installer it starts working again without having to replug or reboot or anything. I'd be happy to provide any logs, etc.

ccMSC, 2016-01-11:
Take a look at #187 and see if the provided binary is any help. If not, try following the instructions to produce a log.

mtbmark, 2016-01-11:
Experiencing the same issue.  On a boot from powered off, keyboard works during post, but once the OS starts to load it dies.  If I unplug the keyboard and plug it back in, it works.  Will try the binary you mentioned tonight and see if that helps.  If not I will gather the logs and upload.  Thanks.

theboasnake, 2016-01-11:
I had the same problem with my K95 RGB, this solved it:
-Don't set ckb as a startup service during installation (remove it if you already set it)
-Start Mac with keyboard in BIOS mode (This may be an extra step, made hackintosh boot way faster)
-Put keyboard back in normal mode and log in
-Start daemon manually
-Open CKB

mtbmark, 2016-01-11:
@theboasnake Thanks for the tip.  I will give that a shot when I get home.

theboasnake, 2016-01-11:
@mtbmark You're welcome!
Let me know if it works.

mtbmark, 2016-01-12:
This is with a Strafe RGB keyboard...

@theboasnake Thanks again for the info.  It seems to be behaving better now.  The key takeaway for me was not having the App or daemon start automatically.  If I wait until Im in OSX, I can start the daemon manually and launch the app.

I'm setup in a dual boot config.  This is a hackintosh and I'm also booting into Windows 10 from time to time.  Using the keyboard in OSX seemed to be more reliable if I took the same approach in Windows and didn't start the Corsair utility automatically.

@ccMSC I tried a few times to capture logs but didn't have any success after modifying the com.ckb.daemon.plist file in the launchdaemons directory.  In fact it seemed like when I added in the log options, the daemon wouldnt start on boot at all.  I could still start it manually after booting.  I also tried the beta binary in issue #187 and didn't seem to have any tangible impact.  Willing to try again if you have any ideas.

Thanks for the help.  At least seem functional for now.

chris37879, 2016-01-13:
Hey, that binary in the other thread solved the issue for me, my keyboard now works on hotplug and reboot.

closed 2016-01-15 by ccMSC