ccMSC/ckb #215: Hide Window on Start-Up

dabfalcon, 2016-01-10:
Keep up the good work on ckb! My problem is that the ckb window appears every startup, which I then have to exit. I am running Manjaro KDE. I thought I was smart and make a startup script launching 'ckb --background' to only start the background process. This worked for the first reboot but not subsequent ones. 

How can I autostart ckb but not have it seen?

ccMSC, 2016-01-11:
There's currently a bug where the session restore causes the window to pop up. You can fix it by going to `System Settings > Startup and Shutdown > Desktop Session` and adding `ckb` to "Applications to be excluded"

dabfalcon, 2016-01-11:
Thanks ccMSC. Legend.

closed 2016-01-15 by ccMSC

diijkstra, 2016-05-07:
Regarding session restore bug, it is because `QGuiApplication` is modifying arguments provided to the Application - when creating `QGuiApplication` instance `-session` param is removed. I think the proper way to check if App is being run during session restore is to use `QGuiApplication::isSessionRestored()`.

See following patch, works for me on Linux + KDE - ckb restored by KDE with `-session xyz` param is hidden on start.

```
diff --git a/src/ckb/main.cpp b/src/ckb/main.cpp
index 98cb2c1..37cb257 100644
--- a/src/ckb/main.cpp
+++ b/src/ckb/main.cpp
@@ -112,7 +112,7 @@ int main(int argc, char *argv[]){
     }
 
     // Launch in background if requested, or if re-launching a previous session
-    bool background = qApp->arguments().contains("--background") || qApp->arguments().contains("-session") || qApp->arguments().contains("--session");
+    bool background = qApp->arguments().contains("--background") || qApp->isSessionRestored();
     if(isRunning(background ? 0 : "Open")){
         printf("ckb is already running. Exiting.\n");
         return 0;

```

ccMSC, 2016-05-10:
@diijkstra bah, didn't know that. Could you send me a pull request for that change, please?