ccMSC/ckb #217: Character loss if macro too large

frickler24, 2016-01-11:
Hi, as mentioned in the forum some days ago, if have the following effect:
If I define a macro, which is more than about 28 chars long, there are sometimes characters lost when pressing the macro key. The longer the macro, the higher the probability to lose chars. The chars lost seem to be the same on each try, but in very long sequences, they are not always the same.
So I thought about a timing problem with the threads.

If did some debugging and found a (very unnice) workaround:
In ckb-daemon path, file input.c, static void **inputupdate_keys[**(usbdevice* kb) is a loop, which does single writes via os_keypress() on each character of the macro. The count of write calls is OK, but the result is not.
So I tested a short sleep after each call:
```c
if(action->rel_x != 0 || action->rel_y != 0)
    os_mousemove(kb, action->rel_x, action->rel_y);
else {
    os_keypress(kb, action->scan, action->down);
    if (a > 200) usleep(100); else if (a > 20) usleep(30);    // Here is the hack
}
```
That works on my system, even with macro definition of more than 2800 chars. So I can live with it.
But I don't know, what's the problem behind. 
And have other systems (slower ore faster ones) different thresholds for the bug?
BTW: The bug sems to be independent from the load on the system. If I run a lot of busy processes, I get the same effect with nearly the same probability.
May be that's an academic problem, but I do not know exactly. So if someone has an idea where to look further...

ccMSC, 2016-01-11:
Assume you're talking about the `macro` command in the daemon?

I haven't actually tested it in a long time so it's possible the command itself is bugged, but it's more likely that the OS has a limit on how many events it will process at once and they just get dropped after a while. Sleeping might be the only reliable way to fix it.

frickler24, 2016-01-11:
@ccMSC yes, the macro cmd in the daemon. 
Today I've tested in on an older system, running on mint 17.1 and got the same effect.
So I close this issue here.

closed 2016-01-11 by frickler24