ccMSC/ckb #219: 2 problems with CKB and K70 RGB

WaltRudyJabsco, 2016-01-12:
First off, amazing work!  This is such a useful project and I really appreciate the work that has gone into it.

I recently got a K70 RGB and installed the CKB software.  I'm running on a new iMac with El Cap.

I noticed that the "tilde" (`/~) key is not working correctly -- it is generating a "§" keystroke.

Also, the "pipe" key above the enter key is incorrectly located on the keyboard diagram (it is to the left of the spacebar) and the key is not illuminating when assigned.

Hope it helps to point this out and hope it can be addressed at some point.

Kind regards

DonutEspresso, 2016-01-12:
You probably have the UK layout enabled, instead of US layout.

WaltRudyJabsco, 2016-01-12:
BRILLIANT!  What a goof I am :)  Fixed both issues... Thanks eternally!

closed 2016-01-15 by ccMSC